Fredrum/chiaki #2: chiaki does not launch post build

domb84, 2020-12-08:
**Environment**
 - OS/Distribution: retropie 4.6
 - Desktop Environment: 
 - Hardware: rpi4

**Describe the bug**
chiaki doesn't launch after building from source on rpi4 (retropie 4.6)

Build script...

`# install dependencies
sudo apt install cmake python3-protobuf protobuf-compiler libopus-dev libsdl2-dev libssl-dev -y

# pull Qt5.12 and install
cd /tmp
wget https://github.com/koendv/qt5-opengl-raspberrypi/releases/download/v5.12.5-1/qt5-opengl-dev_5.12.5_armhf.deb
sudo apt install ./qt5-opengl-dev_5.12.5_armhf.deb -y

# clone rpi branch of chiaki
cd /home/pi
git clone --single-branch --branch rpi01 https://github.com/Fredrum/chiaki.git chiaki
cd chiaki
# can be a bit slow
git submodule update --init

# qt5.12 audio fix
# result is the same with or without these paths adding
echo "set(Qt5Core_DIR "/usr/lib/qt5.12/lib/cmake/Qt53DCore")" >> /home/pi/chiaki/gui/CMakeLists.txt
echo "set(Qt5_DIR "/usr/lib/qt5.12/lib/cmake/Qt5")" >> /home/pi/chiaki/gui/CMakeLists.txt
echo "set(QT_QMAKE_EXECUTABLE "/usr/lib/qt5.12/bin/qmake")" >> /home/pi/chiaki/gui/CMakeLists.txt

# build
mkdir build
cd build
cmake -DCMAKE_PREFIX_PATH=/usr/lib/qt5.12 ..
make

# add custom ports launcher
cat <<EOF > /home/pi/RetroPie/roms/ports/RemotePS4.sh
#!/bin/bash
/home/pi/chiaki/build/gui/chiaki
EOF
chmod +x /home/pi/RetroPie/roms/ports/RemotePS4.sh`

ERROR
`root@retropie:/home/pi/chiaki/build/gui# ./chiaki
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
* failed to add service - already in use?`

also tested using the pi user, error remains

Fredrum, 2020-12-08:
Hi are you the same person that reporting a similar issue on the chat group?
Are you attempting to launch Chiaki from a CLI ?

domb84, 2020-12-08:
Resolved in the end. Doesn't require qt5.12 and requires option -platform egfls.

closed 2020-12-08 by domb84