Fredrum/chiaki #4: Temp basic startup and quit for retropie

Fredrum, 2020-12-13:
**What platform does your feature request apply to?**
Choose one or more: Raspbian and Retropi

**Is your feature request related to a problem? Please describe.**
Need to be able to start up a session and Quit it with non DS4 gamepad or KB/M.

**Describe the solution you'd like**
This is going to be a basic tempt fix until someone has time to make a better solutlion.

**Additional context**
Aiming to just do auto startup with Server/Playstation [0]
and
Hard Chiaki quit using R3 button + Moon/Circle

domb84, 2020-12-13:
This is probably not much use for you as it's bash, but if you build the cli theres a wakeup and discover option that gives you the status. This means you can wakeup the ps4 and launch a session directly with a simple script.

However... chiaki-cli discover -h $ip and chiaki-cli wakeup -h $ip -r $reg hang indefinitely. They work, but the process never ends, so they are still useless in my script as I can't sed/grep the output. I've tried using `timeout 1 chiaki-cli discover -h $ip` but then I don't get the output.

```ip=$ip
host=$(/home/pi/chiaki/build/gui/chiaki -platform minimal list | sed -E "s/^Host:\s(.+$)/\1/")
mac=$(sed -nE "s/^.+ps4_mac=@ByteArray\((.+)\)/\1/p" /home/pi/.config/Chiaki/Chiaki.conf)
reg=$(sed -nE "s/^.+rp_regist_key=@ByteArray\(([[:alnum:]]+).+\)$/\1/p" /home/pi/.config/Chiaki/Chiaki.conf)
state=$(/home/pi/chiaki/build/cli/chiaki-cli discover -h $ip | sed -E "s/^.+State:\s+([[:alnum:]]+)$/\1/")

if [[ "state" == "standby" ]]; then
        /home/pi/chiaki/build/cli/chiaki-cli wakeup -h $ip -r $reg
        sleep 10
        /home/pi/chiaki/build/gui/chiaki -platform eglfs stream $host $ip
        #/home/pi/chiaki/build/gui/chiaki -platform eglfs
elif [[ "state" == "ready" ]]; then
        /home/pi/chiaki/build/gui/chiaki -platform eglfs stream $host $ip
else
        /home/pi/chiaki/build/gui/chiaki -platform eglfs
fi
EOF

```

Fredrum, 2020-12-13:
Have have something that's pretty close to working now.
I will need to enable this Retropie stuff on command line but haven't got there yet.
I'm getting close on the start up and quit otherwise.
Might have something late tonight fingers crossed.

Fredrum, 2020-12-14:
@domb84 
Do you want to try with this temp repo,    https://github.com/Fredrum/retroTest

If this works for you I'll copy it over to the 'rpi01' branch in my chiaki fork.

It works by adding '--retropie' to the command line. (in addition to your '-platform eglfs')
It would be good if you could test and see how it behaves when the PS4 is in sleeping mode(orange).
There's some dialogs popping up and I'm not sure if they will mess things up or not.

Let me know.

domb84, 2020-12-14:
I can't get it to launch on Retropie without '-platform eglfs'. Controller works ok for me with that but I've not tested your verion as yet. I'll build it later and report back.

domb84, 2020-12-14:
> @domb84
> Do you want to try with this temp repo, https://github.com/Fredrum/retroTest
> 
> If this works for you I'll copy it over to the 'rpi01' branch in my chiaki fork.
> 
> It works by adding '--retropie' to the command line. (in addition to your '-platform eglfs')
> It would be good if you could test and see how it behaves when the PS4 is in sleeping mode(orange).
> There's some dialogs popping up and I'm not sure if they will mess things up or not.
> 
> Quit is pressing down the R Analogue stick and 'O' button at the same time.
> 
> EDIT: I'm getting controller problems with '-platform eglfs'. Is that needed?
> 
> Let me know.

What is the --retropie option supposed to do? Does it wake the ps4 and launch the session from the Chiaki.conf file?

domb84, 2020-12-14:
```root@retropie:/home/pi/chiaki/build/gui# ./chiaki --retropie --platform eglfs
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Failed to move cursor on screen HDMI1: -13
Could not set cursor on screen HDMI1: -13
QEGLPlatformContext: Failed to create context: 3009
Unknown option 'retropie'.```

Unknown option?

Fredrum, 2020-12-14:
It should be with two dashes --retropie

domb84, 2020-12-14:
It is?

`root@retropie:/home/pi/chiaki/build/gui# ./chiaki -platform eglfs --retropie
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Failed to move cursor on screen HDMI1: -13
Could not set cursor on screen HDMI1: -13
QEGLPlatformContext: Failed to create context: 3009
Unknown option 'retropie'.
root@retropie:/home/pi/chiaki/build/gui# ./chiaki -platform eglfs -retropie
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Failed to move cursor on screen HDMI1: -13
Could not set cursor on screen HDMI1: -13
QEGLPlatformContext: Failed to create context: 3009
Unknown options: r, e, t, r, o, p, i, e.`

domb84, 2020-12-14:
Did you commit your changes? Last update was 12 days ago.
https://github.com/Fredrum/retroTest/commits/master

Fredrum, 2020-12-15:
Ah yes. Can you try this,
git@github.com:Fredrum/retroTest2.git

domb84, 2020-12-15:
Ok I've rebuilt using this and it very nearly works, although for some reason I receive a black screen but I have audio!

I can see it sends the wakeup packet, then starts the session, but it believes the session has ended. I'll try and record it and sent it over. Here's the log:

```
QEGLPlatformContext: Failed to create context: 3009
[I] Discovery Service detected new host with id F8461C7E45AF
In RetroPie Mode
Triggering wakeup
[I] Discovery Service: Host with id F8461C7E45AF is no longer available
[I] Discovery Service detected new host with id F8461C7E45AF
In RetroPie Mode
Auto starting Session
[I] Logging to file /home/pi/.local/share/Chiaki/Chiaki/log/chiaki_session_2020-12-15_20-35-04-421421.log
[I] Chiaki Version 1.3.0
[I] Raspberry Pi Decoder initialized
[I] Controller 0 opened: "X360 Controller"
[I] Starting session request
[I] Trying to request session from 192.168.0.80:9295
[I] Connected to 192.168.0.80:9295
[I] Sending session request
[I] Session request successful
[I] Starting ctrl
[I] Ctrl connected to 192.168.0.80:9295
[I] Sending ctrl request
[I] Ctrl received http header as response
[I] Ctrl received ctrl request http response
[I] Ctrl connected
[I] Ctrl received Login message: success
[W] Received Ctrl Message with unknown type 0x6
[W] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[W]      0 00 01                                           ..
[W] Received Ctrl Message with unknown type 0x16
[W] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[W]      0 00 00                                           ..
In RetroPie Mode
Auto starting Session
[W] Received presumably invalid Session Id:
[W] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[W]      0 10 49 6e 76 61 6c 69 64 53 65 73 73 69 6f 6e 49 .InvalidSessionI
[W]     10 64                                              d
[I] Ctrl received valid Session Id: InvalidSessionId
[I] Starting Senkusha
[I] Takion connecting (version 7)
[I] Takion enabled Don't Fragment Bit
[I] Takion sent init
[I] Takion received init ack with remote tag 0xb18ccf, outbound streams: 0x64, inbound streams: 0x64
[I] Takion sent cookie
[I] Takion received cookie ack
[I] Takion connected
[I] Senkusha sending big
[I] Senkusha successfully received bang
[I] Senkusha Ping Test with count 10 starting
[I] Senkusha enabled echo
[I] Senkusha sending Ping 0 of test index 0
[I] Senkusha received Pong, RTT = 2.024 ms
[I] Senkusha sending Ping 1 of test index 0
[I] Senkusha received Pong, RTT = 2.899 ms
[I] Senkusha sending Ping 2 of test index 0
[I] Senkusha received Pong, RTT = 2.028 ms
[I] Senkusha sending Ping 3 of test index 0
[I] Senkusha received Pong, RTT = 1.925 ms
[I] Senkusha sending Ping 4 of test index 0
[I] Senkusha received Pong, RTT = 2.193 ms
[I] Senkusha sending Ping 5 of test index 0
[I] Senkusha received Pong, RTT = 2.023 ms
[I] Senkusha sending Ping 6 of test index 0
[I] Senkusha received Pong, RTT = 1.951 ms
[I] Senkusha sending Ping 7 of test index 0
[I] Senkusha received Pong, RTT = 1.863 ms
[I] Senkusha sending Ping 8 of test index 0
[I] Senkusha received Pong, RTT = 1.217 ms
[I] Senkusha sending Ping 9 of test index 0
[I] Senkusha received Pong, RTT = 1.034 ms
[I] Senkusha disabled echo
[I] Senkusha determined average RTT = 1.915 ms
[I] Senkusha starting MTU in test with min 576, max 1454, retries 3, timeout 9 ms
[I] Senkusha MTU request 1454 (min 576, max 1454), id 1, attempt 0
[I] Senkusha MTU 1454 success
[I] Senkusha determined inbound MTU 1454
[I] Senkusha starting MTU out test with min 576, max 1454, retries 3, timeout 9 ms
[I] Senkusha sent initial client MTU command
[I] Senkusha received expected Client MTU Command
[I] Senkusha MTU 1454 out ping attempt 0
[I] Senkusha MTU ping 1454 success
[I] Senkusha determined outbound MTU 1454
[I] Senkusha sending final Client MTU Command
[I] Senkusha is disconnecting
[I] Senkusha closed takion
[I] Senkusha completed successfully
[I] Takion connecting (version 9)
[I] Takion sent init
[I] Takion received init ack with remote tag 0x7033129, outbound streams: 0x64, inbound streams: 0x64
[I] Takion sent cookie
[I] Takion received cookie ack
[I] Takion connected
[I] StreamConnection sending big
```

domb84, 2020-12-15:
BTW the exit function works perfectly so this is so close!

domb84, 2020-12-15:
After a couple more tests it seems to be intermittent, but 90% of the time it works.  For my selfish purposes this is perfect!

If I were to be hyper critical, the discovery when it starts seems to take up to 30 seconds or so. Thats not the case with the standard build. And a configurable exit combination would be amazing, but this works amazingly for me.

Fredrum, 2020-12-15:
@domb84

I'd like to get it better than 90%. Thanks for the log but I don't know all
the systems well enough to be able to tell from that.
It doesn't seem to say 'Session closed'.

The session you posted the log for did that go black but kept sound?
Basically if the stream window loses 'focus' it will drop the image. Which
actually I might be able to change for when in retropie mode.

Are you testing over wifi?  If you are would you be able to test over
Ethernet too? Just to check that its not a network thing?

Fredrum, 2020-12-15:
btw it should absolutely not take 30sec to find find and connect with your
PS4. Iv'e never had that so I wonder what's going on there.

domb84, 2020-12-19:
> @domb84 I'd like to get it better than 90%. Thanks for the log but I don't know all the systems well enough to be able to tell from that. It doesn't seem to say 'Session closed'. The session you [...]

Got a clip at last. The session connects but for some reason it thinks it’s already connected (it’s not). Ok the error and the session is fine. I think it’s just connecting twice after the wake packet has been sent?

https://mega.nz/file/1KIyWJha#LJsqX6sw6p084dMBl7Xb5_SPZqZznDCYGaTcXTCyTDw

https://mega.nz/file/IagWBZZS#uvA12b8JHb1zfFLlU4-aEnCUA4Wn_DUd_18CP7qv4Rs

Fredrum, 2020-12-19:
Hi im setting up a fresh retropie sd card and i have gone through and followed your custom install script.

Im not getting a 'ports' page in emulation station though.

Could something be missing?

Fredrum, 2020-12-20:
Nevermind i installed lzdoom and then the ports page showed up with chiaki in it.

Ill do some testing.

domb84, 2020-12-20:
I changed some paths on my system which I changed back to standard in the script last night. I’ve not tested it yet but it should still work.

Fredrum, 2020-12-26:
I've made some changes and it seems to work pretty well for me.
This version is still in 'retroTest2'. I'll move it over to my rpi01 branch if this works well enough for you.
 
Some notes on how I'm using it though that differs from your scripts,

- I'd suggest using my version of dependency installs you shouldn't need more Qt libs than my list and there's no gstreamer used in this. Audio has worked fine for me.

- You shouldn't hardcode in 720p in your eglfs.json as that would limit 1080p use. 1080p is available on PS4 Pro only so I'm not able to test but some people have told me its working. (although there's a new issue with someone having trouble with it).
I instead added something for the --retropie flag that checks the main monitor resolution and full screens to that. So it should work with whatever the users monitor is. The RPi's hardware display will just scale to fill the resolution and black-border when aspect ratio differs.

- I just need to use this command,    
`chiaki --platform eglfs --retropie`

- I don't need to use any of these:    
```
export QT_QPA_EGLFS_ALWAYS_SET_MODE="28"
export QT_SCALE_FACTOR=2
export QT_QPA_EGLFS_KMS_CONFIG=/opt/retropie/configs/ports/chiaki/eglfs.json
```

- Controller registration doesn't seem to work 100% not sure why yet. Mostly works but seems like sometimes a re-plug is needed.

domb84, 2021-01-03:
I've just managed to test this and the same thing happens where it launches the session, it works fine for a few seconds then  it says "Chiaki session has quit: Remote play on Console is already in use" but you can still here the audio in the background and if you ok the dialogue box the session is fine. Problem I have is that I don't have a mouse attached to the Pi so the pop up means going off to get a mouse, ok'ing the dialogue box and carrying on.

I'm pretty sure it's launching 2 sessions by mistake...

Fredrum, 2021-01-09:
I just added another gating test that should prevent it to try starting a session twice in a row.
In the same 'retroTest2' repo.
Not sure how closely you're following but the main dev has added PS5 functionality to the Master repo and also he fixed the grey/black screens when doing 1080p. That last piece might not be committed yet but possibly by the end of the weekend or something.

Once he's done all that I'm going to make a new version here, adding these retropie hack ins.

domb84, 2021-02-02:
sorry been busy with other things. Will take a look at this now!

domb84, 2021-02-02:
FYI get exactly the same problem...

Wake packet sent...
Session starts...
Dialogue pops up saying Chiaki Session has qui, Remote play on console is already in use
Ok dialogue, session still fine

mongo527, 2021-03-03:
Hey @domb84. Where are you seeing that the retropie option was added on thestr4ng3r source? I don't see it anywhere and also trying to test it out. Thanks!

sergiohf, 2021-04-20:
Anyone knows if i can install chiaki on Retropie? Just did a clean install of Retropie and installed chiaki from Dominic's script: https://dominicbird.com/knowledge-base-2/emulation/compile-chiaki-and-add-as-a-custom-port-to-retropie/ 
But the following error appears: QEGLPlatformContext: Failed to create context: 3009
What am i doing wrong?

sergiohf, 2021-04-20:
Just saw that even with the error: QEGLPlatformContext: Failed to create context: 3009, the program starts, but only in command line. But when i tried to open on top o Attract Mode Frontend, and i received the additional errors:
Could not set DRM mode for screen HDMI1 (Permission denied)
Could not queue DRM page flip on screen HDMI1 (Permission denied)

domb84, 2021-05-08:
> Just saw that even with the error: QEGLPlatformContext: Failed to create context: 3009, the program starts, but only in command line. But when i tried to open on top o Attract Mode Frontend, and i [...]

I've never tested with attract mode, but the script works fine with emulationstation still. However the button mapping is all wrong in the latest releases.

carlosmellado, 2021-06-22:
Hi guys,
trying to implement this on my RetroPie, everything worked, but when running the Chiaki script on Ports, runcommand.log says:

```
Could not open DRM device /dev/dri/card1 (No such file or directory)
Could not open DRM device
/opt/retropie/configs/ports/chiaki/chaiki.sh: line 9:  3561 Aborted
```

This happened to me also with a Debian lite OS so then tried to switch to RetroPie to completely follow your steps, but now it's happening the same.
In fact, I have no /dev/dri folder...

Can you guys help me on this?

Thanks

carlosmellado, 2021-06-22:
Just saw that on running raspi-config>advanced options>GL Driver> I have three options:
1 - Legacy (Which /dev/dri is missing)
2 - GL (Fake KMS) (/dev/dri is there, but there is no card1, just card0. And emulationstation does not launch)
3 - GL (Full KMS) (/dev/dri is there, but there is no card1, just card0. And emulationstation does not launch)

Any help would be appreciated!

Fredrum, 2021-06-22:
When you say "running the Chiaki script" which script are you referring to?

For the driver selection I would just try the two KMS systems. Rebooting
inbetween.
I haven't got my Pi handy so I can't check what Iv'e been using. I know
there's been improvements to the Full KMS  quite recently and before that
it was a bit of a WIP, and might still be?

carlosmellado, 2021-06-22:
Sorry, I'll try to explain better.

I was trying to run Chiaki on my RetroPie (last and fresh version, just installed today), by running a script in Ports emulator. I followed this guide posted on @sergiohf 's message

https://dominicbird.com/knowledge-base-2/emulation/compile-chiaki-and-add-as-a-custom-port-to-retropie/

But then I'm getting the following error when launching chiaki:

```
Could not open DRM device /dev/dri/card1 (No such file or directory)
Could not open DRM device
```

So, I decided to follow the oficial guide here, on this github, because the other guide was compiling chiaki using the master branch, and not the rpi01 branch as you said in your guide, but now when I do it, this time, I get this error:

```
Could not find DRM device!
Aborted
```

That's all... I don't know what I'm doing wrong, or how to manage this. 
But trying to execute chiaki from terminal, or from RetroPie doesn't work for me.

carlosmellado, 2021-06-22:
Well, in fact:
Running `./chiaki` gives me:
```
qt.qpa.screen: QXcbConnection: Could not connect to display 
Could not connect to any X display.
```
And running `./chiaki -platform eglfs` gives me:

```
Could not find DRM device!
Aborted
```

Not sure wich one should I run, but both didn't worked.

carlosmellado, 2021-06-23:
Ok, I was able to run chiaki from terminal/retropie by running:
`./chiaki -platform linuxfb`
but now, my question is… 
how can I select your PlayStation without keyboard/mouse? I have it registered yet, but is there any way to launch chiaki and open my PlayStation automatically, or by using my Bluetooth controller?

Fredrum, 2021-06-24:
Iirc I never got that properly sorted for the non-desktop hacky solution.
For desktop I worked out that I could use the PS4 controllers touchpad as a
mouse and select the registered ps icon and double click it to start.

closed 2021-08-14 by Fredrum