Fredrum/chiaki #7: 1080p produces a grey screen

manderson21, 2020-12-22:
**Environment**
 - OS/Distribution: Raspberry Pi OS
 - Desktop Environment: 
 - Hardware: Raspberry Pi 4b and PS4 Pro
 - Exact Version or Commit Hash and Installation Method: Chiaki Version 1.3.0

**Describe the bug**
When switching resolution to 1080p, the stream connects. I get audio and can control the PS4 pro, however the video is just a grey screen.

Fredrum, 2020-12-26:
Hi,
so you can see the video when 720 is selected in the Chiaki settings?

Could you please paste the shell output from the failing session here?

Kou-kun42, 2020-12-27:
Hi,
I'm having the same issue as well.  I just installed Chiaki on my RaspPi 4B, following your build instructions.  720p works great with my ps4 pro but I'm getting a gray screen on 1080p with a bunch of random black bits scattered throughout.  Here's the shell output from my session:

libEGL warning: DRI2: failed to authenticate
QEGLPlatformContext: Failed to create context: 3009
[I] Discovery Service detected new host with id 2CCC4444B662
[I] Logging to file /home/pi/.local/share/Chiaki/Chiaki/log/chiaki_session_2020-12-27_20-25-19-465465.log
[I] Chiaki Version 1.3.0
[I] Raspberry Pi Decoder initialized
[I] Controller 0 opened: "PS3 Controller"
[I] Starting session request
[I] Trying to request session from 192.168.1.38:9295
[I] Connected to 192.168.1.38:9295
[I] Sending session request
[I] Session request successful
[I] Starting ctrl
[I] Ctrl connected to 192.168.1.38:9295
[I] Sending ctrl request
[I] Ctrl received http header as response
[I] Ctrl received ctrl request http response
[I] Ctrl connected
[I] Ctrl received Login message: success
[W] Received Ctrl Message with unknown type 0x6
[W] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[W]      0 00 01                                           ..              
[I] Ctrl received valid Session Id: 1609100720V6D4GS636HKSTZP23YA7WUSVF2FCP43RNLPVLG4EUEPHWJMVQMZOKHWMZGCVTEX2
[I] Starting Senkusha
[W] Received Ctrl Message with unknown type 0x16
[I] Takion connecting (version 7)
[W] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[I] Takion enabled Don't Fragment Bit
[W]      0 01 ff                                           ..              
[I] Takion sent init
[I] Takion received init ack with remote tag 0x3c9fd38e, outbound streams: 0x64, inbound streams: 0x64
[I] Takion sent cookie
[I] Takion received cookie ack
[I] Takion connected
[I] Senkusha sending big
[I] Senkusha successfully received bang
[I] Senkusha Ping Test with count 10 starting
[I] Senkusha enabled echo
[I] Senkusha sending Ping 0 of test index 0
[I] Senkusha received Pong, RTT = 2.768 ms
[I] Senkusha sending Ping 1 of test index 0
[I] Senkusha received Pong, RTT = 2.651 ms
[I] Senkusha sending Ping 2 of test index 0
[I] Senkusha received Pong, RTT = 7.763 ms
[I] Senkusha sending Ping 3 of test index 0
[I] Senkusha received Pong, RTT = 131.326 ms
[I] Senkusha sending Ping 4 of test index 0
[I] Senkusha received Pong, RTT = 8.379 ms
[I] Senkusha sending Ping 5 of test index 0
[I] Senkusha received Pong, RTT = 1.677 ms
[I] Senkusha sending Ping 6 of test index 0
[I] Senkusha received Pong, RTT = 1.562 ms
[I] Senkusha sending Ping 7 of test index 0
[I] Senkusha received Pong, RTT = 1.727 ms
[I] Senkusha sending Ping 8 of test index 0
[I] Senkusha received Pong, RTT = 1.592 ms
[I] Senkusha sending Ping 9 of test index 0
[I] Senkusha received Pong, RTT = 1.584 ms
[I] Senkusha disabled echo
[I] Senkusha determined average RTT = 16.102 ms
[I] Senkusha starting MTU in test with min 576, max 1454, retries 3, timeout 80 ms
[I] Senkusha MTU request 1454 (min 576, max 1454), id 1, attempt 0
[I] Senkusha MTU 1454 success
[I] Senkusha determined inbound MTU 1454
[I] Senkusha starting MTU out test with min 576, max 1454, retries 3, timeout 80 ms
[I] Senkusha sent initial client MTU command
[I] Senkusha received expected Client MTU Command
[I] Senkusha MTU 1454 out ping attempt 0
[I] Senkusha MTU ping 1454 success
[I] Senkusha determined outbound MTU 1454
[I] Senkusha sending final Client MTU Command
[I] Senkusha is disconnecting
[I] Senkusha closed takion
[I] Senkusha completed successfully
[I] Takion connecting (version 9)
[I] Takion sent init
[I] Takion received init ack with remote tag 0x1f31b697, outbound streams: 0x64, inbound streams: 0x64
[I] Takion sent cookie
[I] Takion received cookie ack
[I] Takion connected
[I] StreamConnection sending big
[I] BANG received
[I] StreamConnection successfully received bang
[I] Crypt has become available. Re-checking MACs of 0 packets
[D] StreamConnection received audio header:
[D] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[D]      0 02 10 00 00 bb 80 00 00 01 e0 00 00 00 01       ..............  
[I] Audio Header:
[I]   channels = 2
[I]   bits = 16
[I]   rate = 48000
[I]   frame size = 480
[I]   unknown = 1
[I] ChiakiOpusDecoder initialized
[I] Audio Device alsa_output.platform-bcm2835_audio.analog-stereo opened with 2 channels @ 48000 Hz, buffer size 23992
[I] Video Profiles:
[I]   0: 1920x1080
[I] StreamConnection successfully received streaminfo
[I] Switched to profile 0, resolution: 1920x1080
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] StreamConnection is disconnecting
[I] StreamConnection sending Disconnect
[I] StreamConnection was requested to stop
[I] StreamConnection closed takion
[I] StreamConnection completed successfully
[I] Ctrl requested to stop
[I] Ctrl stopped
[I] Session has quit

Are the two errors in the beginning something to worry about?
Thanks for your help!

Fredrum, 2020-12-28:
Thanks for the log dump. I can't see anything that's showing me where the problem is.
It looks nice and stable at the last 1/3 where the streaming is happening. So yes since I don't have a PS4 i'm not sure how I can solve this.  One more thing you could do is to make sure you have enough bandwidth for 1080p on the network you are using? Can you stream 1080p60 youtube video?  (which would have a fair bit lower bitrate since its video is pre-encoded and not on-the-fly encoded so it wouldn't be a perfect test)

Kou-kun42, 2020-12-28:
1080p remote play with Chiaki works great on all the other platforms I've tried it on like my android phone, my Ubuntu desktop, and on a windows laptop.  Here's another log from a longer session where there's some warnings about missing units and an FEC failed error:

libEGL warning: DRI2: failed to authenticate
QEGLPlatformContext: Failed to create context: 3009
[I] Discovery Service detected new host with id 2CCC4444B662
[I] Discovery Service: Host with id 2CCC4444B662 is no longer available
[I] Discovery Service detected new host with id 2CCC4444B662
[I] Logging to file /home/pi/.local/share/Chiaki/Chiaki/log/chiaki_session_2020-12-28_18-34-06-305305.log
[I] Chiaki Version 1.3.0
[I] Raspberry Pi Decoder initialized
[I] Controller 0 opened: "PS3 Controller"
[I] Starting session request
[I] Trying to request session from 192.168.1.38:9295
[I] Connected to 192.168.1.38:9295
[I] Sending session request
[I] Session request successful
[I] Starting ctrl
[I] Ctrl connected to 192.168.1.38:9295
[I] Sending ctrl request
[I] Ctrl received http header as response
[I] Ctrl received ctrl request http response
[I] Ctrl connected
[I] Ctrl received Login message: success
[W] Received Ctrl Message with unknown type 0x6
[W] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[W]      0 00 01                                           ..              
[I] Ctrl received valid Session Id: 1609180446RD7UFGZLDQ3GO4RJDCCUN2ZHEYJ6QE2D2QREUKFK5WD5UUVDRHBYY5BGCM5DQNCA
[I] Starting Senkusha
[I] Takion connecting (version 7)
[I] Takion enabled Don't Fragment Bit
[I] Takion sent init
[I] Takion received init ack with remote tag 0xb18ccf, outbound streams: 0x64, inbound streams: 0x64
[I] Takion sent cookie
[I] Takion received cookie ack
[I] Takion connected
[I] Senkusha sending big
[W] Received Ctrl Message with unknown type 0x16
[W] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[W]      0 01 ff                                           ..              
[I] Senkusha successfully received bang
[I] Senkusha Ping Test with count 10 starting
[I] Senkusha enabled echo
[I] Senkusha sending Ping 0 of test index 0
[I] Senkusha received Pong, RTT = 1.727 ms
[I] Senkusha sending Ping 1 of test index 0
[I] Senkusha received Pong, RTT = 1.702 ms
[I] Senkusha sending Ping 2 of test index 0
[I] Senkusha received Pong, RTT = 1.734 ms
[I] Senkusha sending Ping 3 of test index 0
[I] Senkusha received Pong, RTT = 2.885 ms
[I] Senkusha sending Ping 4 of test index 0
[I] Senkusha received Pong, RTT = 1.601 ms
[I] Senkusha sending Ping 5 of test index 0
[I] Senkusha received Pong, RTT = 1.786 ms
[I] Senkusha sending Ping 6 of test index 0
[I] Senkusha received Pong, RTT = 2.160 ms
[I] Senkusha sending Ping 7 of test index 0
[I] Senkusha received Pong, RTT = 1.726 ms
[I] Senkusha sending Ping 8 of test index 0
[I] Senkusha received Pong, RTT = 2.162 ms
[I] Senkusha sending Ping 9 of test index 0
[I] Senkusha received Pong, RTT = 1.626 ms
[I] Senkusha disabled echo
[I] Senkusha determined average RTT = 1.910 ms
[I] Senkusha starting MTU in test with min 576, max 1454, retries 3, timeout 9 ms
[I] Senkusha MTU request 1454 (min 576, max 1454), id 1, attempt 0
[I] Senkusha MTU 1454 success
[I] Senkusha determined inbound MTU 1454
[I] Senkusha starting MTU out test with min 576, max 1454, retries 3, timeout 9 ms
[I] Senkusha sent initial client MTU command
[... 53 lines cut ...]
[D] StreamConnection reporting corrupt frame(s) from 1141 to 1144
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] Ctrl received Heartbeat, sending reply
[I] StreamConnection is disconnecting
[I] StreamConnection sending Disconnect
[I] StreamConnection was requested to stop
[W] Received Ctrl Message with unknown type 0x8050
[W] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[W]      0 00 00 00 00                                     ....            
[I] StreamConnection closed takion
[I] StreamConnection completed successfully
[I] Ctrl requested to stop
[I] Ctrl stopped
[I] Session has quit

Hopefully something in this output helps narrow the issue down.

Fredrum, 2021-01-10:
This issue should be fixed now and 1080p resolution should work.
Please try the latest version!

Kou-kun42, 2021-01-11:
Hi I just tried it out and it works perfectly!  Thanks so much for getting this resolved!

Fredrum, 2021-01-11:
That's great news! :) Thanks for letting me know!
Credits go to the Main dev who figured out the 1080p issue!

closed 2021-01-11 by Fredrum