Fredrum/chiaki #9: [E] Failed to Compile Shader: 0:2(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, 1.00 ES, and 3.00 ES

cgioconda, 2021-02-12:
Getting an error immediately after connection, and a black screen.
[E] Failed to Compile Shader: 0:2(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, 1.00 ES, and 3.00 ES

Compiled successfully on Pi 4 running Ubuntu Desktop 20.10 64bit.

Fredrum, 2021-02-18:
Hello, I'm not able to test on Ubuntu but maybe you can try editing this file:

https://github.com/Fredrum/chiaki/blob/master/gui/src/avopenglwidget.cpp

like this:

change all the,
#version 150 core

to

#version 300 core

cgioconda, 2021-02-18:
Tried that, this is the error now:

[I] OpenGL initialized with version "2.1 Mesa 20.2.6"
[I] Ctrl received Login message: success
[W] Received Ctrl Message with unknown type 0x6
[W] offset 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  0123456789abcdef
[W]      0 00 01                                           ..              
[E] Failed to Compile Shader:
0:8(1): error: No precision specified in this scope for type `vec2'
0:9(1): error: No precision specified in this scope for type `vec4'
0:13(2): error: No precision specified in this scope for type `vec3'
0:17(2): error: No precision specified in this scope for type `vec3'

Fredrum, 2021-02-18:
I can't test too much tonight I'll try some things tomorrow but in the meantime you could try adding this under each of those '#version' statements:

`precision mediump float;`

cgioconda, 2021-02-18:
> precision mediump float;

That did the trick!! Thank you so much!

closed 2021-02-18 by cgioconda