Fredrum/chiaki #10: "qt.qpa.screen: QXcbConnection: Could not connect to display"

BobBob15H3r3, 2021-04-05:
Good Morning:

I am experiencing this issue. I was wondering if you might have a solution. I was wondering if you might know why I would get this error. Thank you so much for all your work already, I cannot wait to get this running. I happened on your RetroPi thread by accident and have been excited to see this in action on my Pi4 8GB ever since! Thank you for your time in advance and I look forward to working with you!

Rocknpools, 2021-04-29:
I have the same issue. Are you on raspberryOS lite?

carlosmellado, 2021-06-21:
Having the same problem. I have the feeling that this only works on Desktop based Linux OSes.
I'm having the same issue trying to make it work on raspberryOS lite (Debian) without Desktop.
It would be nice to make it work just by power on the Raspi or on a custom script inside Retropie, but not opening a Desktop and having to attach a keyboard + mouse.

Any help would be appreciated.

carlosmellado, 2021-06-25:
Hey, I was able to run it from terminal / retropie by running with the next parameters:

`./chiaki -platform linuxfb`

Now, looking forward to enable dualsense support :)