Fredrum/chiaki #12: Dualsense support (PS5 controller)

carlosmellado, 2021-06-25:
**What platform does your feature request apply to?**
Raspberry Pi, RaspberryOS, Retropie, or anything running under RaspberryPi and Debian.

**Is your feature request related to a problem? Please describe.**
Yes, I'm unable to play on chiaki by using a Dualsense (PS5) controller.

**Describe the solution you'd like**
Include the new Official driver for Linux

**Describe alternatives you've considered**
Or include steps to install them manually.

**Additional context**
I managed to install the driver by my own, following the next steps:
1. Clone the driver to any folder:
`git clone https://aur.archlinux.org/hid-playstation-dkms.git hid-playstation`
then, enter on this folder:
**cd hid-playstation**
and compile it:
**make**
after that, copy the generated driver into the module folder:
**sudo cp hid-playstation.ko /lib/modules/$(uname -r)/extra**
and enable it:
**sudo depmod
sudo modprobe hid-playstation**

Now, the dualsense controller works on Chiaki, you can start the session with the touchpad on the remote controller, via USB or Bluetooth, but it doesn't work when you're on the session. 
The controller stops reponding and I'm unable to play anything.

P.S: Via USB, the speaker on the controller works, meanwhile, when connected via Bluetooth it doesn't.

Fredrum, 2021-08-02:
Hello!!
The PS5 Dualsense controller should be supported in theory.
What's happening is that to work around an issue with the Sony controllers we are looking up the guid's.
That should get printed out in the terminal you are starting Chiaki I believe. Can you please look for your GUID there and post it back here!?

You can also compare against this list:
https://github.com/Fredrum/chiaki/blob/master/gui/src/controllermanager.cpp

Fredrum, 2021-08-21:
If anyone is having this problem it might be easier to try the new feature to let you remap controllers. It might help.

https://github.com/Fredrum/chiaki/wiki/Controllers#creating-your-own-gamepad-remapping-file

424C414B45, 2021-08-28:
I've tried @carlosmellado 's method, which did result in the DualSense touchpad working to control the mouse cursor - but Chiaki still did not recognize input from the controller. 

So, I tried @Fredrum 's solution, mapping the controller - but I get to this point and: 
> root@raspberrypi:/home/pi# cd SDL
> root@raspberrypi:/home/pi/SDL# mkdir build; cd build
> root@raspberrypi:/home/pi/SDL/build# ./configure
> bash: ./configure: No such file or directory

Fredrum, 2021-08-28:
Oops my instructions were incorrect.
This step should not be there:  ` mkdir build; cd build`
The ./configure should be done in the root SDL directory.

But the remapping would only solve the problem of button presses working in the wrong way. If you have no input triggering whatsoever then the remapping won't make any difference.

I don't have a PS5 or PS5 controller so am not able to test of fix this.

424C414B45, 2021-08-28:
Bummer

carlosmellado, 2021-08-28:
Hey, on my case, I finally have the Dualsense controller fully working.
The problem is that libsdl2-dev on rpi Debian has the 2.0.10 version and 2.0.14 is required.
So I had to update Debian to bullseye (test version) and that allowed me to upgrade libsdl2-dev to 2.0.14 and it worked like a charm with no more hackings.

424C414B45, 2021-08-28:
@carlosmellado if it's not too much trouble, could you explain what steps to take to accomplish that? I'm a newb, just got the Raspberry to learn and stumbled across Chiaki as a first project.

Fredrum, 2021-08-28:
Hi Carlos, did you try just building SDL2 from source instead of updating to Bullseye?

carlosmellado, 2021-08-29:
Hi Fredrum,
Nope, I didn’t tried it, as I’m not an expert on Linux and I didn’t know how to do it.
So if someone tries it and works, it would be nice if they explains the steps to make it work.

Fredrum, 2021-11-18:
Closing as I head this is now working with the SDL2 that one gets with Bullseye.

closed 2021-11-18 by Fredrum

bmn001, 2022-03-17:
I'm sorry to revive this closed issue, but I'm really struggling to get this working in Ubuntu 21.10.

Kernel is 5.13.0-37, libsdl2-dev is 2.0.14, so this should be working, but it just isn't.  All buttons respond correctly in jstest, so the OS is definitely seeing inputs...but the Chiaki window just will not recognize any button presses.

Any ideas?

carlosmellado, 2022-03-17:
On the Chiaki GUI, AFAIK the only thing that works is the touchpad, that acts like a mouse. So if everything works fine you should see a mouse pointer as soon as you press the touchpad. Then, you can double click on your PlayStation by using it.