Fredrum/chiaki #13: Aspect Ratio Scaling / Fullscreen on non desktop OS

abwezi, 2021-08-06:
What platform does your feature request apply to?
Raspberry Pi 4, running Retropie

Is your feature request related to a problem? Please describe.
I'm able to initiate a stream but my screen is a 4:3 monitor, the high resolution streams stretch off screen and low resolution stream only partially fills the screen but is off center. I appear to have no way to scale the video to fit my screen in an aspect ratio correct way. I attempt to use the F11 key described as fullscreen in the wiki and nothing happens and I don't happen to have a DS4 to attempt trying to double tap the touchpad to try using that hotkey I only have a PS5 and Dualsense controller I'm not sure if it's been implemented for that yet.

Describe the solution you'd like
To have a way to fullscreen the stream automatically for desktopless OS maybe a parameter you could run it with idk.

Describe alternatives you've considered
Not using Retropie.

Additional context
I have a pretty non-standard set up of a Pi 4 running composite to a CRT television so the resolution is limited to 720x480. It would help to know where settings are stored that way maybe I can look at all the options on the Retropie version manually because even the options menu stretches off screen at this resolution lol. [Here's an example of the best picture I've managed to get so far](https://ibb.co/kD6Mmvh)

Fredrum, 2021-08-06:
Hello!
Chiaki for Retropie was never written or tested to run on non-desktop unfortunately.
Having said that I managed to do it with some success but it was a while ago now so I don't remember the details. I'd suggest reading through the 'issue' below called something like Temp Basic Startup....

I wonder if the screen since its an older one, does not report its own resolution back to the RPi hardware the way newer screens do?  Maybe you can find a way of spoofing that?

Or some settings in the /boot/config.txt that deals with screen resolutions?  What have you got there now?

abwezi, 2021-08-06:
I can copy my config.txt but the composite output of the Raspberry Pi 4 is always on mode CEA 6 720x480 and that is almost certainly what is reported to Chiaki and what it is outputting the gui at. The issue is that resolution is too small for any stream above 360p and when 360p mode is set then it does not fill the screen and is offset to the top left like in the picture above. I did peek through that other issue you mentioned. I attempted to append --retropie to my launch script like I saw you describe but then for me it would fail to launch following that and I can't access the test repos that you were using at the time that are mentioned.

Here is what works for me currently to open the app through the ports menu

`#!/bin/bash
/home/pi/chiaki/build/gui/chiaki -platform linuxfb`

Fredrum, 2021-08-06:
I'm not following when you say "that resolution is too small for any stream above 360p and when 360p mode is set".
Where do you set 360p mode?

The resolution of the vide stream from the playstation should be independent from the the resolution of the screen on the client machine. It should get stretched to fit with aspect ratio calculated in.

abwezi, 2021-08-06:
It seems that it should work correctly if I could access the "fullscreen mode" but nothing seems to work on my keyboard or controller. Instead of whatever resolution option I set in the Chiaki working like you described and scaling resolutions even up to 1080p to an aspect ratio correct frame fitting my screen size what happens instead is that a stream window appear that I can not full screen no matter how many times I press F11 and if that stream window is a resolution above 360p than anything greater than my screens resolution will be cropped off screen. [Here is another link to a picture to help illustrate.](https://ibb.co/PgScWKh) With your link I was able to compile the RetroTest build and even successfully run it with the --retropie option it but it appears that build was created before PS5 support was added. Sorry, I'm a novice with this stuff, but would you be able to point me in the right direction towards how to merge your changes in this retrotest build into the newest version with PS5 support?

Fredrum, 2021-08-06:
Ok I see so you're using a PS5.  Yes that's correct that text version was most likely made before the PS5 support was added.
I'd like to help but I can't promise when I can get time to do an update. I also can't promise that if I do it will solve this problem, as I don't think I can emulate your setup.

I always find Git very confusing myself as I'm just a hobbyist programmer so don't want to try to give advice there.
I'll try to take a look maybe over the weekend and see what I can see.

If you're able to find a way to emulate your screen resolutions with just the hdmi out and a regular LCD that would be helpful.

abwezi, 2021-08-06:
Thank you your offer to help is appreciated! Unfortunately I do not have access to a PS4 at this time. I believe if you add the lines
`hdmi_group=1` and `hdmi_mode=6` it will set you to 720x480i 60 hz but I don't think it matters that you use that same exact mode you can probably do `hdmi_mode=2` for 480p instead should work about the same I'd imagine

Fredrum, 2021-08-07:
Hi I have just made a new version in my main repo taking the latest version from the Main dev's master for some general updates and then added my previous --retropie tweaks.

Iv'e done a few tests and it sort of works but with a couple of problems remaining.

I don't have a PS5 so maybe you can try it out by following the standard instructions except for the 'got clone' part.

**Please do:**
`git clone https://github.com/Fredrum/chiaki.git`

instead of
`git clone --single-branch --branch rpi01 https://github.com/Fredrum/chiaki.git`

I'm running from Raspberry CLI command prompt like this:

`~/path_to/chiaki/build/gui/chiaki --platform eglfs --retropie`

**Some problems**:
1. When quitting out from Chiaki there seems to be a crash and a freeze.
2. Sometimes I get a dialog popping up saying "Wakeup Packet Sent". You'll need to use your DS controllers touchpad to click that away if it appears.

**Other Notes**:
1. It should automatically connect to your Playstation. You should not have to double click in the Chiaki gui. At least that works for me on the PS4.
2. When in chiaki and Playstation session press R3+Circle to quit. (and possibly freeze as described above)
3. Make sure to set codec to h264 in the chiaki prefs.

abwezi, 2021-08-07:
This worked perfectly for me in terms of getting the picture. Everything is centered and scaled to the proper aspect ratio now in full screen! However I am running into an issue now that I am attempting to go further I think I saw another user describe. I am using a Wii U Pro Controller via an adapter that reports it as a standard Xbox 360 controller. The buttons seem to work okay in the menu, all face buttons appear to function correctly, but the home button on the controller does nothing, select functions as the PS button, and LT bound to the share button. Then when I launch a game from the menu the face buttons stop functioning and I can no longer press anything but those share and PS buttons to do anything. I think the other user Domb84 was encountering a similar problem. This problem was actually happening for me even before this new build I'm pretty sure so I don't think it has to do with any of the recent changes

Fredrum, 2021-08-07:
Hi great to hear that its working for you with the session stream.
I have not had any problems with either a DS4 or Xbox controllers. You might have to do some general problem shooting for that one.
Try the 'jstest' command/tool if you haven't already.
You have 'joystick' installed already?
There's also some mapping settings in the Chiaki prefs but I haven't ever used those.

closed 2021-08-23 by Fredrum