Fredrum/chiaki #14: Hi great to hear that its working for you with the session stream.

abwezi, 2021-08-07:
What platform does your feature request apply to?
Choose one or more: Raspberry Pi 4 running Retropie

Is your feature request related to a problem? Please describe.
Controls for Chiaki are not bound properly to controller detected as Xbox 360 controller. A certain amount of buttons work until a game is launched for some reason but once a game is started all buttons but the currently bound PS button (Left Trigger) and Share button (Select) stop responding. This appears to have happened to other user @Domb84 as described in https://github.com/Fredrum/chiaki/issues/4 "Theres something wrong with the controller mapping now. I can't actually us any of the square, triangle, circle, x buttons int game (but the work in the main sony menu oddly). Also the right joy stick doesn't work. The previous build was ok." his description seems very similar so this may have been caused by a change implemented sometime around that time between when he had it previously working in other builds Dec 2020 and when he attempted to use your test build. I don't believe this has anything to do with the changes made to allow the --retropie parameter I seemed to have this issue with other previously compiled versions as well. I believe the readme states testing was done with an Xbox One Controller which I think may have different ID's and such and might be why controllers identified as Xbox 360 are having this problem. Not sure

Describe the solution you'd like
Default bindings for Xbox 360 controller to match DS4/5 as much as possible and work once a game is started as well as the menu.

Additional context

I was advised "I have not had any problems with either a DS4 or Xbox controller. You might have to do some general problem solving for that one.
Try the 'jstest' command/tool if you haven't already.
You have 'joystick' installed already?
Are the buttons working in native emulators or games?
There's also some mapping settings in the Chiaki prefs but I haven't ever used those."

_Originally posted by @Fredrum in https://github.com/Fredrum/chiaki/issues/13#issuecomment-894672625_

I can run jtest when I'm home later, I'm not sure how to check if 'joystick' is installed, but I'm not sure if answering your question at the end may help more because yes this current controller I'm using is properly detected as an Xbox 360 controller by Retroarch and several other emulators and applications I've used like Mupen64Plus, Daphne, Steamlink, all with correct default bindings so it definitely appears to be working correctly elsewhere and any problems I'm having using this gamepad on my Pi 4 so far are exclusive to this application here. If you happen to know where I can locate these prefs for Chiaki in order to edit them I'd be happy to look more into that myself. Thank you again you've been greatly helpful!

Fredrum, 2021-08-07:
Hi use your regular apt command to install joystick and jstest.
They might already be installed.

`sudo apt install joystick`

jstest is actually part of the 'joystick' package.  This is a very standard joystick handling suite of tools for linux.

If you google you can find many ways of detecting joypads.
For example do,

`ls /dev/input/js*`

and every return is a joystick device, in theory.

then you can for example do
`jstest /dev/input/js0`

When I do that and press the DS4 'PS' button I get a hit for Button: 10.

```
Driver version is 2.1.0.
Joystick (Sony Computer Entertainment Wireless Controller) has 8 axes (X, Y, Z, Rx, Ry, Rz, Hat0X, Hat0Y)
and 13 buttons (BtnA, BtnB, BtnX, BtnY, BtnTL, BtnTR, BtnTL2, BtnTR2, BtnSelect, BtnStart, BtnMode, BtnThumbL, BtnThumbR).
Testing ... (interrupt to exit)
Axes:  0:     0  1:     0  2:-32767  3:     0  4:     0  5:-32767  6:     0  7:     0 Buttons:  0:off  1:off  2:off  3:off  4:off  5:off  6:off  7:off  8:off  9:off 10:off 11:off 12:off
```

abwezi, 2021-08-07:
Okay so I have ran to install 'joystick' and am told the latest version is currently installed. I will paste a log of showing all detected devices here

`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 045e:028e Microsoft Corp. Xbox360 Controller
Bus 001 Device 003: ID 062a:4101 MosArt Semiconductor Corp. Wireless Keyboard/Mouse
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

[Here is a log of jstest](https://controlc.com/026b5fd2) running with my controller and then me inputting in the following order. Dpad U>D>L>R > Face Buttons Y>A>X>B > Left Stick U>D>L>R>L3 > Right Stick U>D>L>R>R3 > LB>RB > LT>RT > Select>Start>Xbox Home Button

Not sure where to go from here

Fredrum, 2021-08-07:
I can see that you for example getting Button10 triggered which for me is the XBox 'home' button.

I just tried to use my own "Microsoft X-Box One pad (Firmware 2015)" controller and while it looks to work as expected in jstest id doesn't behave that way in Chiaki. Like you suggested.

This worked before so it would take me some time to figure it out I believe.
If you want to get going soon I can only recommend getting maybe a DS4 or PS5 controller.

I will want to get this fixed but not sure when I can do it.

abwezi, 2021-08-07:
I'm not sure if it could be somehow

abwezi, 2021-08-07:
Take your time thanks so much again! Does your keyboard exhibit any of the issues I've described? Because even that has the same issue of all bindings outside of the PS button and share button stop working outside inside of games. I will consider seeing if I can get the Dual Sense controller to work but I asked @carlosmellado who had the earlier issue about Dual Sense support how it was going for him and he seemed to be saying he couldn't get it working on retropie due to the latest libsdl2-dev breaking it

Fredrum, 2021-08-07:
Yes I thought of the SDL2 angle, which Chiaki uses for controller input, and tried with this program   https://github.com/meleu/jstest-sdl

Which I think should be using whatever SDL2 is installed before building it.

Doing that my xbox controller worked as expected.
Which makes me guess that it's not a problem in sdl2.

abwezi, 2021-08-07:
Oh sorry I think I was needing to be more specific with that last comment. I would attempt to use my PS5 controller but I looked at issue https://github.com/Fredrum/chiaki/issues/12 and it appears at the end of the thread that the user @carlosmellado was unable to fully get his PS5 controller working having it only functioning in the gui at that time but not within the remote play session and you provided him after installing the drivers. I reached out to him to inquire if he had figured out how to get the controller fully functioning in stream and all and he had this to say 

> yes! But it’s quite difficult depending on the OS you use. I installed the driver first manually, but you can search it through the web by searching AUR hid-playstation driver. And the latest thing but most important, you should have libsdl2-dev on a version that supports the dualsense controller. Default one on Debian I think is 2.0.9 that doesn’t work, so I had to upgrade the system to bullseye and install the 2.0.14 version.
This way it worked, but it broke Retropie :(

Seems to imply that currently Retropie is unable to support the most recent libsdl2-dev version required to support the PS5 controller here but I'm not sure. I'd like to try but I'm concerned about bricking my OS

Fredrum, 2021-08-07:
Yes that seems to start getting a bit involved and adventurous!
It might be good to get a cheap smaller micro sd card that you can experiment with if you want to continue.

You could try building libsdl yourself and then try linking against that rather than the system installed one but that might send you down the rabbit hole of dependencies.  https://github.com/libsdl-org/SDL

Or see if you can find a way to translate your Wii Pro controller into DS4 input instead for now.

abwezi, 2021-08-14:
Yeah just updating that I haven't had any luck with attempting either of those fixes, I seem to be encountering similar issues trying to get the PS5 controller working and can't find a way to spoof my controller to a PS4 controller like you can on Windows from my searches. In the mean time I've managed to achieve something useable through streaming to the chiaki windows application on my PC and then streaming that to my Raspberry Pi through the SteamLink app. The latency is not great due to the two layers of streaming but it's playable until any progress gets made on this issue. Thanks again.

Fredrum, 2021-08-17:
Could you please send me the GUID of the controller as detected by Chiaki?

It should tell you in the shell output as your starting. Its a long number looking something like this:

030000005e040000dd02000003020000

I might have to do specific re-mappings for different xbox controllers possibly.

abwezi, 2021-08-17:
My log states

> [2021-08-17 12:43:57:919919] [I] Controller 0 opened: "Microsoft X-Box 360 pad (030000005e0400008e02000010010000)"

Fredrum, 2021-08-18:
As things stand the DB currently has a number of mappings that are
incorrect on older kernel versions. The map addressed by this patch is
wrong when the kernel is up to date. I agree a more thorough fix is needed
to handle the disparity but for the time being it seems sensible to me all
mappings work in the modern case as opposed to some being broken in either
case.

https://github.com/libsdl-org/SDL/pull/4344

Fredrum, 2021-08-18:
That's interesting I was thinking about trying that but have been focusing on SDL2 as I was hoping I could solve it without more drivers etc.

I actually managed to get my own XBox One controller mapping working. But I had to run a little program that comes with the libsdl project, and then add to Chiaki the reading of a re-map file created with that program.

There's a big public list of controller guid's and remap info but when I tried it didn't actually work with my xbox pad. The equivalent file I generated myself did work though.

Iv'e just started a discussion with someone involved in that database project.

You can read some of what Iv'e done here,  https://github.com/libsdl-org/SDL/issues/4655

abwezi, 2021-08-18:
Okay great I would be happy to attempt generating the re-map myself just let me know! Do you know particularly what might achieve changing the "select" button on my PS3 controller to act as the "touchpad middle click" of a PS4 controller? Or am I out of luck with the touchpad button without a real PS4 controller?

Fredrum, 2021-08-19:
Alright so lets see if this works then!

Please re-build and install Chiaki from the master ie,
`git clone https://github.com/Fredrum/chiaki.git`

Then I put a copy of the remap software here,
https://github.com/Fredrum/controllermap_builds

I'm not sure if that will run for you though as you had a RPi3 ? Also I'm currently on libsdl higher than 2.0.16.

If it doesn't work you could easily build it yourself. You need to build the main repo,

https://github.com/libsdl-org/SDL

NOTE: I wouldn't run 'install' on it after your build as you might want to keep the Raspian native sdl.

then you also need to build the 'test' apps in,
https://github.com/libsdl-org/SDL/test

you'll need to do

`./configure `
then
`make`

Then run test/controllermap

At the end you'll get a text string which you'll just copy+paste into a textfile you can name whatever you want.

That's the file you want to point to in the new chiaki preferences field I just added called 'Remap File'.
Please use the full path.

If that ends up working for you I'd love it if you could post the working file here and I can show to the people keeping the data base.

abwezi, 2021-08-19:
This worked for me!

The mapping file I generated was the following

030000005e0400008e02000010010000,Microsoft X-Box 360 pad,a:b0,b:b1,x:b2,y:b3,back:b8,guide:b10,start:b9,leftstick:b11,rightstick:b12,leftshoulder:b4,rightshoulder:b5,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,leftx:a0,lefty:a1,rightx:a2,righty:a3,lefttrigger:b6,righttrigger:b7,platform:Linux,

Now the mappings are correct, similar to the PS3 controller.

closed 2021-08-19 by abwezi

Fredrum, 2021-08-21:
I just re-flashed a microsd card with a 10 month old Raspbian image (official) and installed Chiaki on that.
My Xbox One controller is working perfectly on that setup. So something has changed since then. I'm still looking for the precise change though.

Fredrum, 2021-08-21:
Hi Abwesi are you installing chiaki on a Retropie setup!?

I'm not able to get the controlle problem with 'real' Raspbian OS images. Both old ones and new ones work fine with my xbox controller.

So it would be useful to know if you also were using a Retropie image to install Chiaki on!?

abwezi, 2021-08-21:
Yes that is correct I am running on a Retropie setup!

Fredrum, 2021-08-22:
Ok so someone replied to one of my issue tickets and the problem is that Retropie installs a customised version of the kernel driver 'xpad'.

My xbox one controller problems went away after I removed that custom driver.

You can do that in the Retropie Setup  -> Manage Packages -> Manage Driver Packages

find and uninstall 'xpad'.

Apparently that driver is an old fix that might not be needed any more and might not be needed for your controller. See,  https://github.com/RetroPie/RetroPie-Setup/issues/3379

You should be able to uninstall that and then your pad might work fine without the remapping file in the Chiaki prefs window.