Fredrum/chiaki #15: Request: Option to bind "select/back" to the center touchpad click for PS3 / Xbox controllersa game open, after installing Xpadneo. Now all that is needed for this to work properly is the rebinding let me know if there's anything more I can do to help. I was also able to pair my PS3 controller and it appears to work without any bindings being "incorrect" it would essentially be perfect except that the select button is mapped to "share" and since so many games put menus behind the touch pad click it would be preferable if select was that instead otherwise I'm unable to open the map in most games. Things are very close!

abwezi, 2021-08-19:
I was also able to pair my PS3 controller and it appears to work without any bindings being "incorrect" it would essentially be perfect except that the select button is mapped to "share" and since so many games put menus behind the touch pad click it would be preferable if select was that instead otherwise I'm unable to open the map in most games.

_Originally posted by @abwezi in https://github.com/Fredrum/chiaki/issues/14#issuecomment-901326288_

Now that Xbox controllers can be properly remapped in the newest version things are working nearly perfect for me but without some way to swap out the default binding for the select/back button on these controllers from the share button to the often used in games touchpad button the experience will be hampered by needing a keyboard essentially to access many secondary menus in games etc. This would be the last step to make this perfect for me pretty much so anything you can do is really appreciated you've been amazing help

abwezi, 2021-08-19:
This may not be a full fix for the issue as it doesn't really provide an option and may break other controllers that can properly emulate both touchpad and share button but I may be able to make the edit described here: https://github.com/thestr4ng3r/chiaki/issues/169#issuecomment-644482271 and rebuild it will work for my usecase. I will try tomorrow and report back

abwezi, 2021-08-19:
That worked for me and now this is working perfectly for my use! I just applied the described fix in the last post to chiaki/gui/src/controllermanager.cpp changing line

state.buttons |= SDL_GameControllerGetButton(controller, SDL_CONTROLLER_BUTTON_BACK) ? CHIAKI_CONTROLLER_BUTTON_SHARE : 0;

to

state.buttons |= SDL_GameControllerGetButton(controller, SDL_CONTROLLER_BUTTON_BACK) ? CHIAKI_CONTROLLER_BUTTON_TOUCHPAD : 0;

and now select is touchpad button for me! Until there is some sort of other mapping functionality for this assignment this may be the best way and possibly more of an issue with the main fork? Feel free to close this but thanks so much again Fred!

Fredrum, 2021-08-19:
:D ha ok yes I guess that works for you! 
The track pad when used on a PC is as a emulated Mouse. So moving your fingers is moving the mouse and clicking on the pad is clicking the mouse.

You could possibly also try some programs like 'joy2key' or 'qjoypad'. The last one I think needs X11 which you are not using so might not work. I haven't tried joy2key but I believe Retropie might be using it?

You might also be able to write a python script to hack around it.

In case you need that other button back some day.

sbappudi, 2023-06-24:
> That worked for me and now this is working perfectly for my use! I just applied the described fix in the last post to chiaki/gui/src/controllermanager.cpp changing line
>
> state.buttons |= [...]

This no longer seems to work for me using an xbox one controller. I made this change before installing SDL and compiling chiaki. Any other suggestions?