Fredrum/chiaki #18: NO AN ISSUE- Wanted to reach out to the dev and this is the only way i thought of

VGCCodes, 2021-09-01:
Would it be possible to read game data over the stream? Possibly by intercepting packets or something?

Fredrum, 2021-09-01:
Hello!
I'm not actually the main dev on Chiaki but I did a lot of the work adding Raspberry Pi capability and a few other things.
Apart from that, what precisely do you mean by game data?

I think mainly the data being sent from the PS4/5 is video and audio plus some meta data to do with that.
There's feedback to the controllers to make the haptics.

Cheers

VGCCodes, 2021-09-06:
By game data i mean being able to pull the video and use it externally, as when i try to take screenshots the game goes black, so i was wondering if just pulling the game data and making my own screenshot script may be the best route

Fredrum, 2021-09-06:
So you mean that the game stream window goes black when you try to screenshot?
Are you using raspberry pi?  If you are then the window will go black when not 'in focus'. (Due to me having problem figuring out the layering when doing hw decode to screen).
What you might be able to do is maximize the window with F11 or DS4 trackpad double tap, then press 'Print Screen' on a keyboard. That should save an image out to your home directory.

If you want to go the hard route you can re-program in the file 'pidecoder.c' to spit out raw h264 or maybe dump frame pixels. But I'm afraid I won't be able to tell you how.

VGCCodes, 2021-09-11:
Oh i see! Thanks ill try that first solution asap.

VGCCodes, 2021-09-13:
> So you mean that the game stream window goes black when you try to screenshot?
> Are you using raspberry pi? If you are then the window will go black when not 'in focus'. (Due to me having problem [...]

Unfortunately, even when going full screen, when screenshotting the whole image is black
[image]

Fredrum, 2021-09-14:
Oh ok. I guess the hardware 'path' the RPi is using for this decode+display is somehow not accessible to regular screenshots.
I'm afraid that's not something that I'll be trying to do anything about.

closed 2021-11-18 by Fredrum