Fredrum/chiaki #19: Instructions for building Chiaki doesn't work on RPi OS (RPi 4 model B).

llimitt, 2021-11-14:
Would love an updated tutorial how to build (this) Chiaki for the RPi 4 model B+ with Raspberry Pi OS. QT5-default is no longer available = build not possible.

Fredrum, 2021-11-15:
Hi!  Can you tell me what Version of the Raspberry Pi OS you are running?
You can type in terminal,

`uname -a`

And if you could please pate that back here?

chiakakaka, 2021-11-15:
Hi Fredrum, same issue here. Rpi OS is now upgraded to debian bullseye. Same issue with that qt5 message. Everything else seems to install and the gui launches, you can connect, but it's a black screen, nothing is rendered.

Ps - I was impressed and pleasantly surprised how well it ran and I'd pay money to have the ps5 dualsense controller working over bluetooth! I reckon you could sell Pi's with chiaki nicely and simply running for folk. Or even market a "chiaki pi image".  There's a gap in the market for this I say!

Fredrum, 2021-11-15:
Unfortunately I bricked my last remaining Pi4b about a week ago so I currently have no way or testing up fixing issues.
I do have one on order though which should arrive in a few days and then I can try to figure out what's gpoing wrong!

Thanks for the endorsment! :) I don't have a PS5 either but the PS4 controller works well over Bluetooth on the RPi4b.
I'm hoping the new SDL2 version that ships with Bullsye enables the Dualsense.

chiakakaka, 2021-11-16:
Oh no! How do you go about bricking a pi?!  Cool fingers crossed, thanks for your work on this. Let us know when we should try again installing chiaki and i'll report back / with ps5 dualsense working or not too.

PS and now my naivity, is it possible to package up all the install steps into one "executable" ? Again making it massively  more accessible to lots more people?

thanks again for the hard work!

Fredrum, 2021-11-17:
I plugged in 9v because I forgot that I my settable power plug was on that setting!  :P
I have managed to get it to work now on my own Bullseye RPi4b and have edited the install how-to with the new instructions!  
Please if you could try them out and let me know if that worked out for you. Unfortunately you'll need to do all the steps from the beginning as I edited one of the cmake files. If you did the PS4/5 registration you should not need to do that again.

It is possible to bundle dependencies up  and make a sort of packaged distribution. On Linux you can't make real windows style executables where everything is baked into one file. Which is a big drawback on Linux.  I'm not sure if I want to do that though as I'm guessing it'll take more time that I have to give to this project. We'll see. Maybe once I have my 'new generation' version I'll consider it.

chiakakaka, 2021-11-17:
Tested.. It works! ***And so does the dualsense PS5 controller*** ! :)

So now you have a perfect little remote play solution you could make nice and easy for folk.

Fredrum, 2021-11-17:
Cool! :)   Great to hear that the Dualsense is working now too!

chiakakaka, 2021-11-17:
:) Yeap nearly perfect, however realising just now that the touchpad doesn't work! Anything obvious I can do for that issue at this point? :D

Fredrum, 2021-11-17:
Not that I'm aware of.
It shoould work in some fashion even if possible not supporting all use casse features native to a PS.
I'd be interested to learn if its just the PS5 controller or the PS4 too?

chiakakaka, 2021-11-18:
> Not that I'm aware of. It should work in some fashion even if possibly not supporting all use casse features native to a PS. I'd be interested to learn if its just the PS5 controller or the PS4 too?

sadly no DS4 here. Does the trackpad work on the ps4 controller?

llimitt, 2021-11-19:
Sorry for being late to the party, but thanks! The new install instructions work fine!

closed 2021-11-24 by Fredrum