Fredrum/chiaki #21: Sudden exits

Korijn, 2021-12-12:
I'm playing on raspberry pi 4b, remoting to a PS4 Pro. I've set resolution to 720p and everything is on the wired network. Sometimes, chiaki just randomly exits in the middle of a session, and even with verbose logging I can't see anything out of the ordinary.

It seems to have started happening once the PS5 remote play support launched, but I could be wrong on that...

Any ideas?

(I am running a debug build though)

Fredrum, 2021-12-15:
Not sure. Could you please post the output on the terminal sheel from after a crash?

Btw, with a PS4 Pro you should be able to do 1080p60.

Korijn, 2021-12-15:
Yeah I reduced it to see if that would alleviate the problems, but nope. I'll write the log to a file next time I get around to it.

Thanks for the response & awesome work!

Fredrum, 2022-02-23:
Closing for now.
If this is still an issue and you have more information for me to be able to assess feel free to re-open.

closed 2022-02-23 by Fredrum