Fredrum/chiaki #22: Feature Request: Switch for disabling Video and Audio processing?

Testodyne, 2022-01-04:
Would it be possible to add an option to completely omit audio and video decoding, thus handling only the controller commands? I am currently using your Chiaki fork on a (not so powerful) Raspberry Pi 1 Model B together with a Dualshock 4 controller to play PS5 games that normally would only work with a DualSense controller. Unfortunately, the CPU load is 100% all the time (although decoding is set to 360p, 30 frames, H264) causing a heavy Video stutter. Audio just cracks - if there is an output at all. Unfortunately, when I disable the "Raspberry Pi decoder", the controller commands stops working either. Despite the high CPU load, the controller movements seem to work surprisingly well. At least I didn't notice any lag after playing a few hours. In my opinion this setup is excellently suited as a DualSense replacement. So it would be awesome if this setting could be implemented to reduce the constant high CPU load on legacy devices!

Fredrum, 2022-02-03:
Ha interesting use case!
I might not spend any more time on this original 'rpi01' version as I am working on a 'rpi02' version that is going to be more broadly compatible with different Pi OS's.
I'll keep it in mind my new version. Although its using much more CPU so might not work well with the weakest RPi's.

marcelomanzo, 2022-07-19:
@Fredrum I would be interested in sponsoring ($) this feature.

marcelomanzo, 2022-07-19:
I think it would be even better if we could run it without the gui, just command line and using a raspberry pi os lite for example.

> Ha interesting use case! I might not spend any more time on this original 'rpi01' version as I am working on a 'rpi02' version that is going to be more broadly compatible with different Pi OS's. I'll keep it in mind my for new version. Although its using much more CPU so might not work well with the weakest RPi's.