Fredrum/chiaki #23: Skip pin input

Untz64, 2022-01-12:
So, I've got this working on my RPi 4 with RetroPie pretty well, but the one thing I can't get around is the pin entry after waking my console. I currently either have to turn on the console and login before using Chiaki, or have a keyboard plugged in to input the pin. Is there an entry I can put in the config file to save my pin, or any other work around that I'm not finding?

Fredrum, 2022-02-03:
Hi sorry for late reply. You should only need to add the PIN once, when first registering, so I'm a bit confused about what you are seeing.
Once you have your config file in ~/.config/Chiaki/    you should not need to re-register anything.

Untz64, 2022-02-03:
So, I should have clarified better before: this isn't for the registration pin, but rather the security pin that I've set so that my kids don't log into my account. I'm not even sure it's possible to skip, but even having the ability to key the pin in with a controller would help.

Fredrum, 2022-02-04:
Hmm that sounds a bit out of scope for me to be able to help with. Also still not sure what pin you are referring to. A screenshot might help?

Untz64, 2022-02-05:
I think this is more of an upstream problem, but it's just not something that seems to be taken into consideration since most platforms Chiaki runs on offer some form of keyboard input. I've set a login pin on my PS5 so that I don't have to worry about my kids trying to circumvent play time locks and other settings, but that means that whenever I start a session either locally or via remote play, I have to provide that pin. 

I'm not sure how to take screenshots on my Pi, but here's a screenshot from the Windows and Android clients, and one from the official Sony PS Remote Play app for a comparison as to what I'm trying to illustrate. 
[image]
[image]
[image]

Basically, I'm trying to either pre-fill and send what that prompt is asking for via a config file or environment variable, or have some way to input via a controller so I don't have to keep a keyboard plugged in as this is running on a Retropie image.

Untz64, 2022-02-07:
No worries, I already assumed it wasn't a huge issue for people considering the lack of any issues mentioning it in the upstream repo. Thanks anyways!

closed 2022-02-23 by Fredrum