Fredrum/chiaki #24: Not working over internet?

CyberCowboy, 2022-03-23:
Tried starting a discussion, but it wouldn't let me, so bringing it here:

 I'm new to remote play in general.  Should Chiaki be able to work over the internet (not on same Lan as Playstation)?

I know the Windows PS Remote Play app works from a different site, but taking my Raspberry Pi to the same site, it doesn't see the PS5 at all (does when on the same network.) 

Either way, great work.  I've got one working on the same network for my son to play in his room after everyone else goes to bed, for volume.  I'd like to send another with him so he can play at his mom's house if he wants.

Fredrum, 2022-03-23:
Hi yes no problem sending question here! I should probably shut that 'Discussions' section down as I never look there.
Thanks and its always fun to hear how people use this software! :)

As for 'over the internet' use that's the next thing I'll look into but it will probably we a little while as always for me. Real work, family and other projects takes time too. :) But I have a brother with an PS5 on the other side of the planet and I was going to start figuring that out with him. So far I failed at the first step, making a small rpi disc image that he can just download and burn. He's only got a Pi3b atm so it'd take ages for him to compile Mesa.

But it's no.1 on my ToDo list.

CyberCowboy, 2022-03-24:
The pi my son is using is either a 2 or 3.  I can grab an image of it and put it somewhere for you if you want.     I'm happy to help however I can.

Fredrum, 2022-03-25:
Thanks for the offer. I found a method that seems to work.
Feel free to subscribe to new 'rpi02 updates' thread to get notified when the feature is ready.

Fredrum, 2022-05-04:
Hi I have a version now that is able to connect over the internet.
It's not a super smooth experience doing it but it seems to work at least back to my own PS4.
If you're able to it would be great if you could help test it?

Basically you'll have to do these things:

git clone --recursive https://github.com/Fredrum/remoteTest.git

Build as previously described for my rpi02 branch.

Copy over your Chiaki_rpi.conf to the Other sites /home/pi/.config/Chiaki
Rename it       `your_new_name.remote`  

You'll have to 'Port Forward'  9295-9295 towards your Playstation console.
The first port 9295 (at least) need both TCP and UDP open.

Get your External IP address for where the Playstation console is located.

Type that into the new IPv4 fields.

Make sure that the Playstatoin console is Awake

Fredrum, 2022-05-19:
I have just pushed out a new 'rpi02' version that has the internet connection feature!
https://github.com/Fredrum/chiaki/wiki/More-rpi02-Instructions-and-tips#remote-internet-play

It'd be great to hear if it works for you because it does for me but not for a user with somewhat unconventional networking setup.

Fredrum, 2022-05-20:
Closing this.
You can just open a new issue if you're having trouble with the internet remoting.

closed 2022-05-20 by Fredrum