Fredrum/chiaki #27: Blank screen on fresh installation on pi 4

angelflames1337, 2022-04-28:
Installing it on fresh raspbian 4 bulls eye. And no matter what combination of resolution, codec and decoder, I always get blank screen, with some errors ([image]) keep running in the terminal. Please help, I bought an overpriced Pi 4 specifically to use Chiaki.

Let me know if you need any more info.

Fredrum, 2022-04-30:
Hi I don't own a PS5 on my own. But I have tested with my brother who does and also just have an older Pi3b+ so it should definitely work some way with your Pi4b!

My brother didn't test hevc/h265 so I can't confirm that 100% though it should work in theory.

Have you tried this combination,
h264
720p
60fps

Could you please also paste what's printing out in the shell when you first launch!?

I also belive that when you decode hevc/h265 you might have to add to the CMA memory.
You can do this by editing your /boot/config.txt

dtoverlay=vc4-kms-v3d,cma-512
or maybe
dtoverlay=vc4-kms-v3d,cma-256

Would you please also try that and let me know?

angelflames1337, 2022-05-10:
Hi, 

I tried the combo that you mentioned and it surprisingly work this time! 

Tried the CMA settings you mentioned, and it didn't work out and h265 fails with same error.

Fredrum, 2022-05-10:
Ah well at least you can play now that's good.
Yeah its tricky as I don't have my own PS5 to test with.

Would you mind trying 1080p with the other working settings?

Fredrum, 2022-05-16:
Actually for hevc/h265 I just remembered, maybe you have to do this:

add the following line to your /boot/config.txt
dtoverlay=rpivid-v4l2

meldarionqeusse, 2022-05-17:
I am getting this issue as well with the PS5 with H265 I have added both `dtoverlay=rpivid-v4l2` and `dtoverlay=vc4-kms-v3d,cma-512`. With H264 it works but it laggy. Running on an rpi4 2gb

Fredrum, 2022-05-17:
@meldarionqeusse   
Which version of the operating system are you using? I'm wondering if you don't have the new avcodec but that should be there by default in the latest Bullseye RPi OS.

could you please paste the output of
`uname -a`

and also run 
```
sudo apt update
sudo apt upgrade
```
if you haven't for a while.  (I don't need the output of that)

meldarionqeusse, 2022-05-18:
I just flash the sd card yesterday and did sudo apt update & upgrade before I tried installing Chiaki.

[image]
```
$ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

meldarionqeusse, 2022-05-18:
Yes sure take your time the work and research that you've done is good enough already and especially documentation was clear and easy to follow along. I wanted to re-open the issue just in case others are in the same situation. I will be looking forward to try it out. Thank you for your work and effort

Fredrum, 2022-06-05:
@meldarionqeusse 

Hey Iv'e just added a ffpeg install script that Iv'e made to get Chiaki to work on Ubuntu and I'm hoping it will work on other distro's like Debian too.

You can try to run this,
https://github.com/Fredrum/chiaki/blob/rpi02/install_ffmpeg.sh

If you do would you be able to please let me know how it went?

meldarionqeusse, 2022-06-05:
Hi Thanks for the update. The good news is that it worked. 

However I am getting this error after I rebooted the rpi. It happened before so I removed chiaki-rpi and reinstalled everything but ideally I do not have to repeat the installatio process every time

```
Linux base Path:  /home/pi/Downloads/chiaki-rpi/build/rpi/chiaki-rpi;
SDL version linked: 2.0.20
Segmentation fault
```

meldarionqeusse, 2022-06-05:
Never mind the segmentation fault seems to have fixed itself I think the problem was cause I was running the command from tmux. It works flawlessly. Thanks a lot for your work and for replying. I think the issue can be closed

Fredrum, 2022-06-05:
Great to hear it works on Debian now.
Closing.

closed 2022-06-05 by Fredrum