Fredrum/chiaki #29: Hardware decode rasp 400

rennard, 2022-05-07:
Hey, absolutely loving the software by the way, works perfectly! Just wondering if I'm able to use the hardware decode method on the raspberry pi 400 and if there's a recommended setting? Thanks!

Fredrum, 2022-05-08:
Hi Thanks for the compliment! 
Yes it should definitely use hw decode by default. You'd be able to tell because sw decode is slow and tops out the cpu.
Which version of Chiaki are you using? Are you using any of my rpi01 or rpi02 branches?

Which build+install instructions did you follow?

Which OS are you using on your 400?

rennard, 2022-05-08:
It's running fairly well so I'm assuming it is using it, I was just confused because the drop-down for it was empty by default and I wasn't sure if I should select one.I followed the instructions on your rpi01 branch, and I'm using the default raspian OS. As I say, it's running well I was just wondering if there was anything I could be doing/settings to change to squeeze more performance out!Cheers

Fredrum, 2022-05-08:
Hm,ok I think the rpi01 version should feel better than 'fairly well' so not sure what you are experiencing you might have to describe in more detail.
rpi01 has the lower latency of the two I think and should definitely feel snappy to play through. Best latency is by using cable for both network and controller but its pretty decent with bluetooth controller too.

Fredrum, 2022-05-08:
Oh yeah about the dropdown...there's a checkbox for Rpi decoder. I forget the label but it should say something like that.
So that should be ticked. I don't think the menu is meant to work for the rpi's. iirc

Fredrum, 2022-05-16:
Closing this one.
Keep in mind I'm not updating the 'rpi01' branch any longer. Feel free to try rpi02.

closed 2022-05-16 by Fredrum

rennard, 2022-05-17:
sorry for the late reply, thanks for the info, I was largely wondering if the dropdown worked so no worries!