Fredrum/chiaki #30: Manual adding on RPi02

raigeiki55, 2022-05-16:
Is there a way I could add my ps4 or ps5 manually on RPI02? My raspberry pi os is in 64bit, unless you could update the RPi01, I'm unable to use it since, I needed to connect it remotely from home.

Fredrum, 2022-05-16:
Are you asking about connecting remotely over the internet?
If so, you're in luck as I'm very close to have a version that does exactly that. Maybe you could even test it out for me?
I'll upload something after work in about 10-12h and you could try that before I make it official.

raigeiki55, 2022-05-16:
Sure go for it, upload it so I could try, Thanks!

Fredrum, 2022-05-17:
Ok I just tested the build and connected back to my own PS4 from a rented Pi4 on the other side of the planet and it found and connected correctly. So at least with a PS4 it should work now in theory.

Please follow the standard 'rpi02' instructions except for the git clone line where you instead should do:
`git clone --recursive --single-branch --branch dev_rpi02 https://github.com/Fredrum/chiaki.git chiaki-rpi_dev`

You don't have to register again if you have already done that.

Then follow the new instructions here,
https://github.com/Fredrum/chiaki/wiki/More-rpi02-Instructions-and-tips#remote-internet-play

There's some glitchy bugs still that I need to fix so if anything weird happens just close Chiaki and start again.
Also I'm going to correct the widget layout in the gui.

Let me know how it goes!

raigeiki55, 2022-05-17:
ok will try, gonna give you some feedback sooner or later, thanks

raigeiki55, 2022-05-17:
Hello, I've encountered this:

Will try to read rpi settings file: /home/pi/.config/Chiaki/Chiaki_rpi.conf
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Aborted

Didn't have any issues with building it but don't know if its my monitor's resolution.

How can I fix this? My Resolution is 1368x768 which I can't adjust it. Thanks

Fredrum, 2022-05-17:
Hi thanks for the feedback.
No its not your monitor I think that's probably the program looking for a file that doesn't exist or something like that.
Could you please paste the output of a 'ls' in that folder?  /home/pi/.config/Chiaki/

Does the folder exist? What files are in there?

Fredrum, 2022-05-17:
Yep I just tested here and its crashing when it can't find the file,
` /home/pi/.config/Chiaki/Chiaki_rpi.conf`

I'll need to fix that but in the mean time you could make a copy of the .remote file and name it back so that it matches the .conf file its looking for.

So basically you'll have two files with different names but exactly the same content.

That should get you past the crash.

raigeiki55, 2022-05-17:
well that's fine, but since this is a fresh installation, I don't have it,  should I wait for your fix then, since I don't know where are the files I will copy

Fredrum, 2022-05-17:
No you'll need to have the original /home/pi/.config/Chiaki/Chiaki_rpi.conf from an rpi that was setup up to play locally with your playstation. Can you do that?
This method relies on you having that initial register file that gets created when you register a local (LAN) playstation.

That's the file that you then bring to your remote location and rename as described.

raigeiki55, 2022-05-17:
well I can't but my configs are in my phone where I use chiaki on my android, but anyway I'll just wait till you fix the crash, Hope you keep this thread until you fix it, I'll test it and hope it works, thanks

Fredrum, 2022-05-17:
Ah well these are the Raspberry Pi versions of Chiaki and they are a bit different from the Android one. The .conf files are completely different. Even after I fix the crash thing it still will work in the same way and you will need to use at least one Raspberry Pi.

If you don't have a raspberry pi you'll need to find a Forum or something that can help with the Android version.

Fredrum, 2022-05-17:
Wait in your OP you said you have a Raspberry. So I'm a bit confused now.

raigeiki55, 2022-05-17:
I do, i have a .json file that can export config from android and well I think the raspberry version doesn't have this, so I  will just register the chiaki for raspberry manually

raigeiki55, 2022-05-17:
let's just ignore the android part, and focus with the raspberry, I just wait for your update, rebuild a new version and try it out myself

raigeiki55, 2022-05-17:
> Hi thanks for the feedback. No its not your monitor I think that's probably the program looking for a file that doesn't exist or something like that. Could you please paste the output of a 'ls' in that folder? /home/pi/.config/Chiaki/
> 
> Does the folder exist? What files are in there?

well the folder doesn't exist

raigeiki55, 2022-05-17:
Update: manage it to work by copy paste the config but don't know how to register ps4 or ps5
[image]

raigeiki55, 2022-05-17:
[image]

Fredrum, 2022-05-18:
Basically the process currently is:

1. Set up Chiaki on a Raspberry Pi in a location where you also have a Playstation locally on the same LAN.
2. Register the Playstation following the instructions.   https://git.sr.ht/~thestr4ng3r/chiaki#usage
3. Use the same Raspberry or a different one somewhere else together with the Remote setup descriptions to enable remote play.

raigeiki55, 2022-05-18:
ok will do it later, will send the results soon as possible

raigeiki55, 2022-05-18:
well just made it work indirectly, it now works on local but trying it remotely doesn't, it cannot find the ps4 and the only way i could make it connect remotely is by connecting it first locally then changing it to lte quickly without disconnecting it

Fredrum, 2022-05-18:
Ok so we're making some progress that's good.
Still confused you'll have to try to explain your complete setup here. What is the hardwares + OS's involved at all points?

How and on what are you using 'LTE'?

I have only tried on LAN, Wifi and regular Internet connection without VPN.

raigeiki55, 2022-05-18:
now my setup involves raspberry pi 4 with raspberry pi os bullseye 64bit, since I do not have fiber internet at our store, I'm just using mobile data in order to connect.

The reason I need a tunnel vpn is that I have a cgnat Internet, where my public ip address and WAN ip address are different.

With a tunnel vpn, i can use 192.168.2.1 anywhere with my mobile data including chiaki.

raigeiki55, 2022-05-18:
yup, anyway if I try building the application on a mobile data it will say unknown even with an existing configuration although I tried it locally it worked

raigeiki55, 2022-05-18:
I just hope you could update soon

Fredrum, 2022-05-18:
I don't have any more updates planned now for the networking.
Iv'e been fixing bugs on 'dev_rpi02' and tonight in some hours I will copy that over to the regular 'rpi02' branch.

raigeiki55, 2022-05-18:
okay will do, i'll try your rpi02 update soon, thanks

Fredrum, 2022-05-19:
I have just pushed latest updates to 'rpi02'

raigeiki55, 2022-05-19:
ok thanks will try it again

raigeiki55, 2022-05-19:
[image]
[image]

Still doesn't work, maybe gonna give up on this, rpi01 works but with a black screen, but since there are incompatibilities due it is only 32bit compatible I have no choice but to use rpi02, my only feedback for this is that it won't detect the ps4 on the rpi02, I hope you could help me on this one. Thanks

Fredrum, 2022-05-19:
Yeah sorry I couldn't help you figure it out. I have tested it working from a Raspberry Pi4b in Europe over to my PS4 here in North America. So with a more standard networking setup it seems to work.

raigeiki55, 2022-05-19:
If you have a youtube demo for it, i will use it as a guide in order so i can make it work on my raspberry pi