Fredrum/chiaki #31: Bullseye 32-bit and 64-bit

hybotics, 2022-05-23:
Chiaki does not support ARM 6-bit OS such as Raspberry Pi Bullseye.

Is this on the radar for development?

Fredrum, 2022-05-24:
Yes my 'rpi02' branch should work with 64bit Raspberry OS Bullseye. See,
https://youtu.be/qDthHmD30us?t=39

https://github.com/Fredrum/chiaki/wiki/rpi02---branch-for-Bullseye,-64bit,-h265-and-kmsdrm

Fredrum, 2022-05-26:
Closing this. Feel free to start a new issue if you're having trouble getting 64bit bullseye to work.

hybotics, 2022-05-26:
Why was this closed? Your software will still not build on Bullseye. It does not find SDL even though it has been installed using your sript.

Fredrum, 2022-05-26:
If you want some help you'll have to provide some information on what exactly your doing and what error messages you're having. 

Rpi Hardware.
OS distribution.

hybotics, 2022-10-11:
It can not FIND SDL. I have installed SDL by running the script in the
repository.