Fredrum/chiaki #33: Unable to open stream

JeremyLJordan, 2022-06-05:
Device: RPi 4 B+
OS: Raspbian Bullseye 5.15.32-v7l+
Chiaki Version 2

I am having an issue getting my stream to start and i feel confident it is due to something with the change in dtoverlay. I have followed the tutorial verbatum and did troubleshooting by changing dtoverlay between: 

1. vc4-kms-v3d 
2. vc4-fkms-v3d
3. rpivid-v4l2 (which i know is the correct one for my model)

I get a segmentation fault on both 2 and 3 that outputs to the terminal like so:

`Linux base Path:  /home/pi/chiaki-rpi/build/rpi/chiaki-rpi'�Ĭ���
SDL version linked: 2.0.20
Disp Mode: 	1920x1080px @ 0hz 
SDL_VIDEO_DRIVER selected: x11
Segmentation fault
`

 I can launch the UI using number 1, but when the stream tries to open I get the following

`Linux base Path:  /home/pi/chiaki-rpi/build/rpi/chiaki-rpi���L��
SDL version linked: 2.0.20
Disp Mode: 	1920x1080px @ 60hz 
SDL_VIDEO_DRIVER selected: x11
SDL_RENDER_DRIVER selected: opengles2
Running under X11: 1
INFO: Audio device 0: Built-in Audio Analog Stereo
INFO: Audio device 1: Built-in Audio Digital Stereo (HDMI)
Rpi IO created
END HostInit
Gamepads currently attached: 0
Check the gamepad is properly connected
Will try to read rpi settings file: /home/pi/.config/Chiaki/Chiaki_rpi.conf
Found settings for 1 host(s)
Discovery Service started
GL_VERSION  : OpenGL ES 3.1 Mesa 20.3.5
GL_RENDERER : V3D 4.2
Gui init done
Discovered Host State:  ready
Discovery ID match
Host Click
Starting Play session!
SDL drm_fd:  33554460
Segmentation fault
`
I attempted to revert to chiaki v1, but I am unable to register my device because there is no longer a PIN when registering.

Any help would be appreciated.

Fredrum, 2022-06-05:
Hi! Thanks for adding the Info on what system you are trying to get to work!

- I would always go with `vc4-kms-v3d` these days fkms is older now and will get less and less support.

- `rpivid-v4l2`  is in addition to any of the other two and has no real connection with them. It enables the /dev/video19 hevc hardware decode device for h265 video streaming supported by the PS5 only.

So I'm focusing on your second larger block of output.
I just tried downloading and building again on my RPiOS sd card and it worked with that si we should try to see what you're trying to do that's different.

I see you have `5.15.32-v7l+`  which I believe means 32bit. Is that correct? It should work w 32 bit though.

Do you have a PS4 or PS5?

Fredrum, 2022-06-05:
When I start a stream I get a valid looking _SDL drm_fd:_ which for some reason you don't.

My successful stream startup looks like this,

```
Found settings for 1 host(s)
Gui init done
Gamepad added
Controller is:  PS4 Controller
Host Click
Starting Play session!
SDL drm_fd:  85
[h264_v4l2m2m @ 0x5596288b00] level=-99
[h264_v4l2m2m @ 0x5596288b00] Using device /dev/video10
[h264_v4l2m2m @ 0x5596288b00] driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[h264_v4l2m2m @ 0x5596288b00] requesting formats: output=H264 capture=YU12
Actual Fmt: 181
Finish InitFFmpeg
END chiaki session start
Init Audio CB
ConnectionEventCB() CHIAKI_EVENT_CONNECTED
Starting Play Thread
Connected Setsu Motion Device /dev/input/event2
Connected Setsu Touchpad Device /dev/input/event0
```

I would have to flash a new SD card to see if I can replicate your error.

Fredrum, 2022-06-05:
Arr I found a bug/mistake that  thought I had fixed but apparently not.

Could you please try in your chiaki-rpi folder do,
`git pull`

then do the make again?
You should not need to re-register.

if not you'll need to re-do the 'git clone' step then the build steps.
(but not register)

JeremyLJordan, 2022-06-05:
Hey, thanks for responding so quickly. Its 32 bit. I can try 64 if you think that will change anything. I also attempted another build after installing the FFmpeg script that was recently added. It hasn't helped. I noticed something happening in the build that may help too. 

`/home/pi/chiaki-rpi2/chiaki/rpi/src/drmprime_out.c: In function ‘drmprime_out_new’:
/home/pi/chiaki-rpi2/chiaki/rpi/src/drmprime_out.c:563:1: warning: label ‘fail_free’ defined but not used [-Wunused-label]
  563 | fail_free:
      | ^~~~~~~~~
/home/pi/chiaki-rpi2/chiaki/rpi/src/drmprime_out.c:527:17: warning: unused variable ‘drm_module’ [-Wunused-variable]
  527 |     const char *drm_module = DRM_MODULE;
      |                 ^~~~~~~~~~
[ 88%] Building CXX object rpi/CMakeFiles/chiaki-rpi.dir/src/main.cpp.o
In file included from /home/pi/chiaki-rpi2/chiaki/rpi/src/io.cpp:3:
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h: In function ‘int stbi_write_bmp_to_func(void (*)(void*, void*, int), void*, int, int, int, const void*)’:
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:514:32: warning: missing initializer for member ‘stbi__write_context::context’ [-Wmissing-field-initializers]
  514 |    stbi__write_context s = { 0 };
      |                                ^
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:514:32: warning: missing initializer for member ‘stbi__write_context::buffer’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:514:32: warning: missing initializer for member ‘stbi__write_context::buf_used’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h: In function ‘int stbi_write_bmp(const char*, int, int, int, const void*)’:
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:522:32: warning: missing initializer for member ‘stbi__write_context::context’ [-Wmissing-field-initializers]
  522 |    stbi__write_context s = { 0 };
      |                                ^
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:522:32: warning: missing initializer for member ‘stbi__write_context::buffer’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:522:32: warning: missing initializer for member ‘stbi__write_context::buf_used’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h: In function ‘int stbi_write_tga_to_func(void (*)(void*, void*, int), void*, int, int, int, const void*)’:
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:613:32: warning: missing initializer for member ‘stbi__write_context::context’ [-Wmissing-field-initializers]
  613 |    stbi__write_context s = { 0 };
      |                                ^
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:613:32: warning: missing initializer for member ‘stbi__write_context::buffer’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:613:32: warning: missing initializer for member ‘stbi__write_context::buf_used’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h: In function ‘int stbi_write_tga(const char*, int, int, int, const void*)’:
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:621:32: warning: missing initializer for member ‘stbi__write_context::context’ [-Wmissing-field-initializers]
  621 |    stbi__write_context s = { 0 };
      |                                ^
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:621:32: warning: missing initializer for member ‘stbi__write_context::buffer’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:621:32: warning: missing initializer for member ‘stbi__write_context::buf_used’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h: In function ‘int stbi_write_hdr_to_func(void (*)(void*, void*, int), void*, int, int, int, const float*)’:
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:789:32: warning: missing initializer for member ‘stbi__write_context::context’ [-Wmissing-field-initializers]
  789 |    stbi__write_context s = { 0 };
      |                                ^
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:789:32: warning: missing initializer for member ‘stbi__write_context::buffer’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:789:32: warning: missing initializer for member ‘stbi__write_context::buf_used’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h: In function ‘int stbi_write_hdr(const char*, int, int, int, const float*)’:
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:796:32: warning: missing initializer for member ‘stbi__write_context::context’ [-Wmissing-field-initializers]
  796 |    stbi__write_context s = { 0 };
      |                                ^
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:796:32: warning: missing initializer for member ‘stbi__write_context::buffer’ [-Wmissing-field-initializers]
/home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/stb_image_write.h:796:32: warning: missing initializer for member ‘stbi__write_context::buf_used’ [-Wmissing-field-initializers]
In file included from /home/pi/chiaki-rpi2/chiaki/rpi/include/rpi/host.h:15,
[... cut ...]
/home/pi/chiaki-rpi2/chiaki/rpi/src/io.cpp:603:9: warning: unused variable ‘size’ [-Wunused-variable]
  603 |  size_t size = av_image_get_buffer_size((AVPixelFormat)AV_PIX_FMT_RGB24, outWidth, outHeight, 1);
      |         ^~~~
`

Just to add a little more information, am using the raspberry pi flasher application, I have ssh and vnc enabled for configuration purposes. I also crashed and burned attempting to get this working on a retro pie install. 

can you tell me exactly what version of raspbian you are using and I will use that and try again? 

Thanks.

JeremyLJordan, 2022-06-05:
> Arr I found a bug/mistake that thought I had fixed but apparently not.
> 
> Could you please try in your chiaki-rpi folder do, `git pull`
> 
> then do the make again? You should not need to re-register.
> 
> if not you'll need to re-do the 'git clone' step then the build steps. (but not register)

doing this now. standby

JeremyLJordan, 2022-06-05:
That didn't seem to change anything. I also attempted to resolve this issue after my first attempt, where should this point to? 

error: XDG_RUNTIME_DIR not set in the environment.

I'll wipe the sd card and try again, what version of raspbian are you using with the version? I'll just use that

JeremyLJordan, 2022-06-05:
> Do you have a PS4 or PS5?

PS5

Fredrum, 2022-06-05:
Hm I have just now tried on both 32 and 64 bit regular RPiOS Bullseye's.
Apart from the fix I just did, which was caused by bad resolve of absent rpivid-v4l2 device, it woked on both for me. I only have an PS4 to test with though but It worked not long ago with my brothers PS5. 

I wouldn't bother re-flashing the OS if you've got a standard Raspberry OS install.

The new error message where does that show up?
`error: XDG_RUNTIME_DIR not set in the environment`

Have you got the standard Desktop? Window manager etc?

You could also try starting chiaki just from the tty/cli screen. That would tell us a bit more if it works or doesn't threre too.

JeremyLJordan, 2022-06-05:
> Hm I have just now tried on both 32 and 64 bit regular RPiOS Bullseye's.
> Apart from the fix I just did, which was caused by bad resolve of absent rpivid-v4l2 device, it woked on both for me. I only [...]

Im going to reflash to start from scratch so I can provide you with better answers.

How do you launch from cli?  I never could figure that out. I didn't see it in the tutorial. Ideally, that's what I'm wanting to do. This pi's only purpose is chiaki

Fredrum, 2022-06-05:
To start from the tty, if you are currently on the Desktop,

1.  Press Ctrl + Alt + F3
2. cd to chiaki/build
3. Run as usual,  rpi/chiaki-rpi

JeremyLJordan, 2022-06-05:
I rebuilt with the lite version (console only) and followed all of the steps. I was able to launch it for cli. I copied my Config file over ssh and the segmentation fault happened again. It appears the directory /.config/Chiaki/ is not being created with the install. Might be something to look at. 
I deleted the config file and got the segmentation fault still, so I'm going to try re-registering it with a fresh build. I'll let you know.

Fredrum, 2022-06-05:
If you still get the crash after that I will try again on a fresh RaspiOS card.

And just to check you are using these instructions now?

https://github.com/Fredrum/chiaki/wiki/rpi02---branch-for-Bullseye,-64bit,-h265-and-kmsdrm

JeremyLJordan, 2022-06-05:
> If you still get the crash after that I will try again on a fresh RaspiOS card.
> 
> And just to check you are using these instructions now?
> 
> https://github.com/Fredrum/chiaki/wiki/rpi02---branch-for-Bullseye,-64bit,-h265-and-kmsdrm

yes, I'm almost done rebuilding. Again, I switched to the lite version. The only dependency that I had to manually add was "git".

JeremyLJordan, 2022-06-05:
for clarification with the ps5, does "dtoverlay=" need to be added to the config or does it need to replace the entry that is already there (vc4-kms-v3d)? so should it be 
`
dtoverlay=vc4-kms-v3d
dtoverlay=rpivid-v4l2
`

or 

`
dtoverlay=vc4-kms-v3d, rpivid-v4l2
`
or just

`
dtoverlay=rpivid-v4l2

`

Fredrum, 2022-06-05:
If you want to use h265, which you probably do, your config.txt should have something looking like,

dtoverlay=vc4-kms-v3d
dtoverlay=rpivid-v4l2

JeremyLJordan, 2022-06-05:
Okay, and I've verified that the config file is not being created. I tried to reregister and got the following error. I do have the GUI up from the cli though

`
Host Click
Opening Register Dialog
AccID entered: (REDACTED)==
Regist Target:  1000100
PIN int:  32
Host Addr:  192.168.10.13
END RegistClick
In  host RegistEventCB
Registration fail
`

Fredrum, 2022-06-05:
Hm so you are getting this error from the tty/cli register?

But you managed to register previously from the Desktop, is that correct?

Just to tackle one problem at a time if you can I would use your previous register .conf file if you feel that one worked?

You should have seen a icon with a PS5 image and a blue or orange light.

JeremyLJordan, 2022-06-05:
Yes, the error is from the cli. When I dropped my previous config it gives the segmentation fault, but it did work for a second in the past. 

Should I manually add the /.config/chiaki/ folder and chown it to pi?

JeremyLJordan, 2022-06-05:
I'm wondering if the registration fail error is due to not being able to write to the config file since the folder doesnt exist.

Fredrum, 2022-06-05:
There hasn't been a problem with the folder when registering a PS4, so far.
The folder gets created when it needs to be. But if one is copying then one would have to create the folder.

it needs to have a big 'C' in Chiaki

I am now going to flash a Lite 64bit Bullseye RPiOS card an install from scratch.

JeremyLJordan, 2022-06-05:
Thanks, let me know. My internet just went down so I won't able to test
anything for a bit.

Fredrum, 2022-06-05:
So I just flashed a fresh Raspberry OS 64bit Bullseye 'Lite' card.
Followed the instructions.
Registered again. (ie didn't copy old regist)

And it worked fine.  :P

So not sure what's different for you apart from using a PS5.
Apart from triple checking your PSN Account ID and PIN I don't know what to do here since I don't have my own PS5. The only thing would be to see if my brother could do another test but I can't say when that could happen.

I noticed you had 10 in your PS5 ip address third group.
Is there anything non-straight forward with your networking setup? Between your RPi and your PS5?

JeremyLJordan, 2022-06-06:
Thank you for trying. I'll keep playing with it. The network isn't crazy, just a switch and router. I must be doing something wrong if it worked for you. 

Thanks for keeping this repo up!

Fredrum, 2022-06-06:
I'll try and see if my brother is up for another test and I'll post back if that happens!

JeremyLJordan, 2022-06-06:
I think my config file was corrupt or there was an encoding problem on my end. I was able to recreate it in the terminal and now I am able to get to the GUI continuously. I'm still not getting output on the screen though. It just goes to a black screen

Jeramo, 2022-06-06:
I am having some issues with Chiaki for Raspberry Pi 4b, 

**Linux base Path:  /home/pi/chiaki-rpi/build/rpi/chiaki-rpi���
                                                             ���
SDL version linked: 2.0.20
Disp Mode: 	4096x2160px @ 60hz 
SDL_VIDEO_DRIVER selected: x11
SDL_RENDER_DRIVER selected: opengles2
Running under X11: 1
INFO: Audio device 0: Inbyggt ljud Analog Stereo
INFO: Audio device 1: Inbyggt ljud Digital Stereo (HDMI)
Rpi IO created
END HostInit
Gamepads currently attached: 0
Check the gamepad is properly connected
Will try to read rpi settings file: /home/pi/.config/Chiaki/Chiaki_rpi.conf
Found settings for 1 host(s)
Discovery Service started
GL_VERSION  : OpenGL ES 3.1 Mesa 20.3.5
GL_RENDERER : V3D 4.2
Gui init done
Discovered Host State:  ready
Discovery ID match**

I tried all different settings, such as:
Video Decoder: Automatic
Video Codec: both H264 and H265
Resolutio: both 720 and 1080p
Framerate: 60

And I set the config.txt to:
dtoverlay=vc4-kms-v3d
dtoverlay=rpivid-v4l2

However, it seems to run fine for half a second, then it all crashes. I tried to set the graphics memory to 128 MB but to no avail. I am connected through Ethernet.

Something that might be good to know is that I had to create the path for the Chiaki_rpi.conf-file by my self since the application didn't do that.

Fredrum, 2022-06-09:
Hi You could try again now if you want.
My latest bug fixes should hopefully have helped a lot.

I'd suggest to use not a 'Lite' Raspberr OS image. You can switch to tty playing on any of them later.

If you want to try, re git clone + re-build + probably re-register.

JeremyLJordan, 2022-06-11:
> Hi You could try again now if you want.
> My latest bug fixes should hopefully have helped a lot.
> 
> I'd suggest to use not a 'Lite' Raspberr OS image. You can switch to tty playing on any of them later.
> 
> If you want to try, re git clone + re-build + probably re-register.

I'll do That this weekend. What was the issue?

Fredrum, 2022-06-11:
I think it might be these,
https://github.com/Fredrum/chiaki/issues/34#issuecomment-1149430123

There might still be problems with h265 and I didn't get enough info from the other user to figure that out.
So if you do test, would you please be able to try both h264 and h265?

JeremyLJordan, 2022-06-11:
Sure, I'm going to do it now. Can you tell me what you want me to put in my config file so I don't do it incorrectly and set you back?

Fredrum, 2022-06-11:
You shouldn't have to do anything by hand with the config file.
In the GUI start with,

h264
720
60
[Select an Audio Device]

JeremyLJordan, 2022-06-11:
Auto gave me this

AVERROR(EAGAIN)
[hevc @ 0x55a078a280] Invalid NAL unit 32, skipping.

V4L2, h264, 720, 60, HDMI- Audio is working great with the exception of audio. I'll troubleshoot that and see if I can fix it. still throwing a few of these in the beginning though.

AVERROR(EAGAIN)
AVERROR(EAGAIN)
AVERROR(EAGAIN)
AVERROR(EAGAIN)
AVERROR(EAGAIN)
AVERROR(EAGAIN)

V4L2, h265, 720, 60, HDMI audio gives the same as auto

AVERROR(EAGAIN)
[hevc @ 0x55a078a280] Invalid NAL unit 32, skipping.

JeremyLJordan, 2022-06-11:
I've tried every audio output and cant get sound. but its streaming great other than that.

Fredrum, 2022-06-11:
Cool thanks.
The AVERROR(EAGAIN) in the beginning is normal. It's the first packets before you have a full frame.

So No Audio at all?  Did you get any Audio devices in the bottom pop-up menu

Fredrum, 2022-06-11:
Ok. I will have to wait until I get time on Saturday evening and I will try and see if I get audio.
I currently have it not connected.  Thanks for testing and great that we're getting closer.

JeremyLJordan, 2022-06-11:
for sure. Thanks again. Let me know if you want me to try anything else. 

i was able to get it to work on 1080

JeremyLJordan, 2022-06-15:
If I could get sound on lite my issue would mostly be resolved. While I still have the desktop version still flashed are there some logs or anything I can help with your efforts in h265?

Fredrum, 2022-06-15:
Hi yes for h265 we/you could try running with the debugger 'gdb' and that could give me some hints, like this:

`gdb rpi/chiaki-rpi`
then enter command
`run`
then after the crash do this command for backtrace
`bt
`
Then paste all of the output here.

For the sound I really have no idea how to test that further. Iv'e done a few more test runs here with my Pi4b and PS4 and I get sound every time.  One thing I could try is actually to test the sound output on my 'Lite' os sd card as last I tried is was the full fat OS.

I'll take a look again and see if I can add some more printouts that could help.

JeremyLJordan, 2022-06-15:
The sound issue is not with chiaki it's definitely an issue with raspbian on the pi4 so unless you just want to solve that problem to help, its not really your issue but I appreciate you either way. 

I'll output the logs for you tomorrow and update you.

Fredrum, 2022-06-15:
Ok thank you, maybe I don't know at this point. I wish I could get hold of a PS5 at retail price but I just can't justify to buy at scalp prices. :P

Thanks, before you start tomorrow, make sure to do a,
`git pull`
To get the latest fixes I did the other day.

I just tried with my 'Lite' card install and the sound was working. So unless I have something wrong in my code that breaks it with a PS5...

Fredrum, 2022-06-15:
actually, ontop of the gdb debug check there's another thing we could do. 
If you un-comment this line with the debug,  https://github.com/Fredrum/chiaki/blob/2d4c3f6d08e807b567ad070c2a70dc7c4f648c0b/rpi/src/io.cpp#L435

the re-build and run normally you'll get a big pile of extra ffmpeg output. It's possible we could read something out of that.

JeremyLJordan, 2022-07-05:
Sorry for the delay

pi@raspberrypi:~/chiaki-rpi/build $ gdb rpi/chiaki-rpi
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rpi/chiaki-rpi...
(gdb) run
Starting program: /home/pi/chiaki-rpi/build/rpi/chiaki-rpi
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Linux base Path:  /home/pi/chiaki-rpi/build/rpi/chiaki-rpi
WARNING:  h265 Decoder Not Detected
SDL version linked: 2.0.20

Program received signal SIGSEGV, Segmentation fault.
strcmp () at ../sysdeps/aarch64/strcmp.S:64
64      ../sysdeps/aarch64/strcmp.S: No such file or directory.
(gdb) bt
#0  strcmp () at ../sysdeps/aarch64/strcmp.S:64
#1  0x00000055555a49cc in main() () at /home/pi/chiaki-rpi/rpi/src/main.cpp:67
(gdb)

Fredrum, 2022-07-05:
Hi so with my new version I'm spotting the new error message I added,
`WARNING: h265 Decoder Not Detected`
Which should mean one of two things.

1. You are not using a Rpi4b, as that's the only model supporting h265/hevc decode.
2. You have a RPi4b but have not added correctly to your config.txt the line,
`dtoverlay=rpivid-v4l2`

my code basically checks for the decode device /dev/video19

you could replicate my check by in a shell doing,
`ls /dev/video*`

and see if you have video19

JeremyLJordan, 2022-07-05:
I realized I hadn't edited my config file. You caught it before I edited my post. also I forgot to add the backtrace which is here

(gdb) bt
#0  strcmp () at ../sysdeps/aarch64/strcmp.S:64
#1  0x00000055555a49cc in main() () at /home/pi/chiaki-rpi/rpi/src/main.cpp:67

Fredrum, 2022-07-05:
Hm thanks that was less output than I'd hoped for. But seems to be something with a strcmp so that's something.
I wonder if the` ../sysdeps/aarch64/strcmp.S: No such file or directory.` is a clue as that's not something that's part of Chiaki.

So is it crashing early now? Do you not get to the gui?

I thought it'd be good to try and do the  `av_log_set_level(AV_LOG_DEBUG); ` that I mentioned above but we can't of course if its now crashing even before you can start a session.

Fredrum, 2022-07-06:
ooof I'm able to properly look a this now after work.
I'm also getting that crash now...

Fredrum, 2022-07-10:
Ok it seems to work again. At least it does for me.
I did re-run **install_sdl.sh** but I'm not certain if that was part of it or if something got fixed back in RPi OS (which I still think was the cause).

JeremyLJordan, 2022-07-10:
> ooof I'm able to properly look a this now after work. I'm also getting that crash now...
> 
> UPDATE: Not trying to shift the blame here :) but it seems like its actually a recent update to something in Raspberry OS that is now causing problems with a call to SDL_GetCurrentVideoDriver().
> 
> I tested with an older sd card I had that's not been updated for a while and Chiaki works correctly on that. Iv'e put a question to the Raspberry forum so will have to wait and see if anyone knows.

What version, I'll install that and try.

JeremyLJordan, 2022-07-10:
> Hm thanks that was less output than I'd hoped for. But seems to be something with a strcmp so that's something. I wonder if the` ../sysdeps/aarch64/strcmp.S: No such file or directory.` is a clue as that's not something that's part of Chiaki.
> 
> If you search your system for that file does it not exist?
> 
> ```
> cd /
> sudo find -name strcmp.S
> ```
> 
> So is it crashing early now? Do you not get to the gui?
> 
> I thought it'd be good to try and do the `av_log_set_level(AV_LOG_DEBUG);` (not gdb, just normal run) that I mentioned above but we can't of course if its now crashing even before you can start a session.

pi@raspberrypi:/ $ sudo find -name strcmp.S
find: ‘./run/user/1000/gvfs’: Permission denied

Fredrum, 2022-07-11:
So I would rerun install_sdl.sh.
Then to make sure, re make chiaki.

Then we should be back to before this latest errors, I think.
Worked for me. Let me know.

Fredrum, 2022-09-04:
I just came back from a trip to the other side of the planet where I could test and fix Chiaki for PS5.
The latest current code should now work with PS5 and also h265.

There's still an issue when using h264+tty/cli that causes some video glitching that I need to take a look at and try to fix.

JeremyLJordan, 2022-09-04:
You're the best. I'll check it out. Thank you

JeremyLJordan, 2022-09-05:
Hey, I'm trying to figure out something from one of these pages here

https://github.com/Fredrum/chiaki/wiki/More-rpi02-Instructions-and-tips#ubuntu-and-other-distros

For retro pie you have 

"add /usr/lib"

and

"add /usr/local/lib"

I'm not sure what that means as "add" is not a thing. Can you clarify?

Fredrum, 2022-09-05:
It means to add the that text to the files mentioned.
For example adding the text:   /usr/local/lib

JeremyLJordan, 2022-09-05:
works great! Thanks for fixing it. I have 2 issues I'd like to mention though, and one request.

When at the chiaki main screen it flickers a lot. As soon as I launch the remote play it doesn't do that, so not a big deal.

When in the remote play session, there is a mouse cursor (i dont have one connected) in the top left side of the screen, it never goes away. Also not a huge deal

My request is a little guidance with launching chiaki at boot. I tried creating a script but it segmentation faults because its running it in the background. This pi is only going to be used for this project, any suggestions for getting it to launch at boot? The script I made for connecting the controller automatically works great.

Thanks again

brantai, 2022-09-05:
If you're using the SDL option (that is, not inside an X session) you can make a simple systemd service file like

```
[Unit]
Description=Chiaki
After=network.target

[Service]
ExecStart=<full path to chiaki-rpi> 2>&1
Restart=always

[Install]
WantedBy=default.target
```

in $HOME/.local/share/systemd/user/chiaki.service, run `systemctl --user enable chiaki` to have it launch at user login.

JeremyLJordan, 2022-09-07:
> 

Thanks for your reply. A few things, I dont have a /systemd/user directory. I attempted to follow these directions

https://www.shubhamdipt.com/blog/how-to-create-a-systemd-service-in-linux/

but it seems to crash. Here is what is in my file:
`
[Unit]
Description=Launches Chiaki at boot and restarts if it crashes

[Service]
User=root
WorkingDirectory=/home/pi/chiaki-rpi/build/rpi/
ExecStart=chiaki-rpi
Restart=always

[Install]
WantedBy=multi-user.target
`

Here is the crash:
`● chiaki.service - Launches Chiaki at boot and restarts if it crashes
     Loaded: loaded (/etc/systemd/system/chiaki.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-09-07 02:18:45 BST; 2s ago
    Process: 839 ExecStart=chiaki-rpi (code=exited, status=203/EXEC)
   Main PID: 839 (code=exited, status=203/EXEC)
        CPU: 2ms

Sep 07 02:18:45 raspberrypi systemd[1]: chiaki.service: Scheduled restart job, restart counter is
Sep 07 02:18:45 raspberrypi systemd[1]: Stopped Launches Chiaki at boot and restarts if it crashe
Sep 07 02:18:45 raspberrypi systemd[1]: chiaki.service: Start request repeated too quickly.
Sep 07 02:18:45 raspberrypi systemd[1]: chiaki.service: Failed with result 'exit-code'.
`

Any ideas would be appreciated.

JeremyLJordan, 2022-09-08:
Got this all resolved. the blinking stopped and the cursor went away after I created the service.

Thanks for all of your hard work

closed 2022-09-08 by JeremyLJordan

JeremyLJordan, 2022-10-11:
Sure, can you tell me what you want my config file to look like exactly so
I make sure it's right?