Fredrum/chiaki #34: Stream works for a slight moment, then crashes

Jeramo, 2022-06-06:
I am having some issues with Chiaki for Raspberry Pi 4b,

Linux base Path: /home/pi/chiaki-rpi/build/rpi/chiaki-rpi���
���
SDL version linked: 2.0.20
Disp Mode: 4096x2160px @ 60hz
SDL_VIDEO_DRIVER selected: x11
SDL_RENDER_DRIVER selected: opengles2
Running under X11: 1
INFO: Audio device 0: Inbyggt ljud Analog Stereo
INFO: Audio device 1: Inbyggt ljud Digital Stereo (HDMI)
Rpi IO created
END HostInit
Gamepads currently attached: 0
Check the gamepad is properly connected
Will try to read rpi settings file: /home/pi/.config/Chiaki/Chiaki_rpi.conf
Found settings for 1 host(s)
Discovery Service started
GL_VERSION : OpenGL ES 3.1 Mesa 20.3.5
GL_RENDERER : V3D 4.2
Gui init done
Discovered Host State: ready
Discovery ID match

I tried all different settings, such as:
Video Decoder: Automatic
Video Codec: both H264 and H265
Resolutio: both 720 and 1080p
Framerate: 60

And I set the config.txt to:
dtoverlay=vc4-kms-v3d
dtoverlay=rpivid-v4l2

However, it seems to run fine for half a second, one time it worked for maybe 30 seconds, then it all crashes. I tried to set the graphics memory to 128 MB but to no avail. I am connected through Ethernet.

Something that might be good to know is that I had to create the path for the Chiaki_rpi.conf-file by my self since the application didn't do that.

Jeramo, 2022-06-06:
I'm on a PS5, by the way!

Fredrum, 2022-06-06:
Hi thanks for all the info that's great!

Copy that about the .config path. I'll have to check that out.

So to confirm when you're saying it works for a short while, do you mean that you can play a game for a short while?
Or you only get the Chiaki Gui for a short while and don't get to play?

Can you please try raising your CMA memory allocation?
https://github.com/Fredrum/chiaki/wiki/More-rpi02-Instructions-and-tips#set-cma-memory-split

Lets start with h264 as I know 100% that should work.
Also we should do 720p to start with as that's safer.
After/if we get that working we can raise the specs.

Jeramo, 2022-06-07:
I did what you suggested, and rebooted the system. All right, I did try with H264 (and H265, just to be safe). In 720p, 60 fps.

This is the error message I'm getting, and also a black screen:
**[hevc @ 0x55b6968520] Invalid NAL unit 32, skipping.
[hevc @ 0x55b6968520] Invalid NAL unit 32, skipping.
[hevc @ 0x55b6968520] Invalid NAL unit 32, skipping.
[hevc @ 0x55b6968520] Invalid NAL unit 32, skipping.
[hevc @ 0x55b6968520] Invalid NAL unit 32, skipping.
[hevc @ 0x55b6968520] Invalid NAL unit 32, skipping.
[hevc @ 0x55b6968520] Invalid NAL unit 32, skipping.
[hevc @ 0x55b6968520] Invalid NAL unit 32, skipping.
AVERROR(EAGAIN)**

Fredrum, 2022-06-07:
Ok thanks. Those messages say that its trying to decode h265 (hevc). 

Can you please paste the output from when you use both h264 and h265 settings?

Jeramo, 2022-06-07:
Funny you're mentioning it, because I recieve it regardless if I choose H264 or H265. 

**H264:**
[hevc @ 0x55875d8f20] Invalid NAL unit 32, skipping.
[hevc @ 0x55875d8f20] Invalid NAL unit 32, skipping.
AVERROR(EAGAIN)

**H265:**
[hevc @ 0x559e5f9f20] Invalid NAL unit 32, skipping.
[hevc @ 0x559e5f9f20] Invalid NAL unit 32, skipping.
AVERROR(EAGAIN)

What I also noticed is that, regardless of setting I change through the UI, the Chiaki-rpi.conf-file never changes.
**It stays this:**

version: 1

regist_hosts:
  - host : 1
    isPS5 : 1
    nick : PS5 Vardagsrum
    id : ****REMOVED THIS INTENTIONALLY*****
    remote_ip : 0.0.0.0
    rp_key : ****REMOVED THIS INTENTIONALLY*****
    regist : ****REMOVED THIS INTENTIONALLY*****
    session : 
      decoder : automatic
      codec : automatic
      resolution : 720
      fps : 60
      audio_device : hdmi

If I manually change the codec parameter to H264, it runs. Performance is honestly quite choppy with Ethernet, but it runs. Occasionally getting the AVERROR(EAGAIN) message. I tried to change the resolution parameters to both 720 and 1080, they both work "equally".

Fredrum, 2022-06-07:
Ok that's great. Well, good progress :)
I'll taka a look tonight. I did change something about the video codec selection recently which must have made my logic illogical.

The stutteriness is not good though and that might not be se same issue. It should be really smooth.

Did you add the 
`hdmi_enable_4kp60=1`
to your config.txt   ?

Are you running this from the Desktop?

Could you please try under the 'tty'  (ie the login terminal screen)

Ctrl+Alt+F3
should take you to the black screen.
then run from there.

Jeramo, 2022-06-07:
Actually, I turned off teamviewer and it worked like magic, so that might have been the issue :)
There is some noticeable latency than if I use Remote Play on any other device, and the PS5 controller has "free passage" to the Raspberry Pi, is there anything I can do to lower the latency?

Fredrum, 2022-06-07:
Cool!  :)
I'll still take a look at the codec selection tonight. 

Or maybe the problem is more that it's not writing to the config file? 
Does any setting alter the contents of the config text file?

Hard to tell what you're experiencing in terms of latency. It should be 'good' but not 'amazing'.
Maybe 3-4 frames?  The best latency is when using cables for both network and gamepad.
I believe that playing under the tty could have lower latency, but I'm not 100% on that.

The other branch, rpi012, uses the old raspberry display and graphics stack and actaully has better latency. That one worked like this last I measured,
https://www.youtube.com/watch?v=85vrekEdEVU

Fredrum, 2022-06-07:
Someone said that latency might be lower with h265 so you could test that?
I'm not sure about that though.

Jeramo, 2022-06-07:
1. Nope, nothing is altered if I change stuff through the UI, except for the account configuration stuff.
2. It is OK, I will try to use it under tty and get back to you later.

I tried with H265 but then I unfortunately get a black screen...

Fredrum, 2022-06-07:
Good that it works for a little bit though.
Have you tried increasing the cma?

Like this in the config.txt
`dtoverlay=vc4-kms-v3d,cma-256`

(must be have no spaces)

Jeramo, 2022-06-07:
> Good that it works for a little bit though. Have you tried increasing the cma memory allocation?
> 
> Like this in the /boot/config.txt `dtoverlay=vc4-kms-v3d,cma-256`
> 
> (must be have no spaces) needs reboot

Yep! And also hdmi_enable_4kp60=1, still got that error message though.

Fredrum, 2022-06-08:
Ok I'm hoping Iv'e now fixed the following bugs,
- config file not writing on new register.
- config file not refresh-writing when changing settings in Gui.
- some other bugs related to Settings/config.

So yes you can now either do a `git pull` or completely re clone the git repo.
If you then do
`git log`
you should at the top see
```
Date:   Tue Jun 7 20:37:37 2022 -0700

    fix settings read after regist
```

Then build again.

I'm hoping that you should now be able to change settings and at least stream using h264?

I'd be very interested to hear about the h265 problems and the above mentioned debug procedure might produce some hints.

Playing from the tty should give you a smoother experience possibly with lower latency.

Fredrum, 2022-09-04:
I just came back from a trip to the other side of the planet where I could test and fix Chiaki for PS5.
The latest current code should now work with PS5 and also h265.

There's still an issue when using h264+tty/cli that causes some video glitching that I need to take a look at and try to fix.