Fredrum/chiaki #35: Couldn't open settings file?

UnluckyForSome, 2022-06-30:
Hi! I have a situation where it seems no config file ever gets created, and every time I load the app I have to re-register my console. When the console connects I get an `AVERROR (EAGAIN)` but i'm hoping that's because it's not loading anything out of the settings file.

Below is my output, note how it's trying to load a settings file which I don't think exists?

pi@chiaki:~/chiaki-rpi/build $ sudo rpi/chiaki-rpi
Linux base Path:  /home/pi/chiaki-rpi/build/rpi/chiaki-rpi▒▒▒\▒▒▒
WARNING:  h265 Decoder Not Detected
SDL version linked: 2.0.20
error: XDG_RUNTIME_DIR not set in the environment.
Disp Mode: 1920x1080 @ 60hz
SDL_VIDEO_DRIVER selected: KMSDRM
SDL_RENDER_DRIVER selected: opengles2
Running under X11: 0
INFO: Audio device 0: vc4-hdmi, MAI PCM i2s-hifi-0
Rpi IO created
END HostInit
Gamepads currently attached: 0
Check the gamepad is properly connected
Will try to read rpi settings file: /home/root/.config/Chiaki/Chiaki_rpi.conf
Couldn't open the settings file
Discovery Service started
GL_VERSION  : OpenGL ES 2.0 Mesa 20.3.5
GL_RENDERER : VC4 V3D 2.1
Discovered Host State:  ready
No Hosts found in file
Could not open directory: /home/root/.config/Chiaki
Gui init done

Fredrum, 2022-07-03:
Hmm ok are you running this as root?

Do you have a folder like,
/home/root/.config/
?

If not how much of that path do you have?

Fredrum, 2022-07-03:
I also see that you're getting an error,
`error: XDG_RUNTIME_DIR not set in the environment`

I've never tried this with the older Pi Zero W so not sure if it has to do with that or just something else.
I haven't seen that error before.

troed, 2023-07-30:
I might have a similar issue. I did create the .config/Chiaki folder manually after seeing the error, but there's still no settings file created whatever I do. All file/folder permissions are correct.

I do have both a PS4 Pro and PS5 on the network, if that's cause for concern. I'm unable to register any of them, it seems when I try to register the PS5 Chiaki tries using the PS4 method and vice versa ... but all that could be the results of the settings file I guess.

(I don't have the XDG_RUNTIME_DIR comment)

> Linux base Path:  /home/pi/chiaki-rpi/build/rpi/chiaki-rpi���pi��
> h265 Hardware Decoder Detected
> SDL version linked: 2.0.22
> Running in Desktop mode:  0
> Disp Mode: 1920x1080 @ 60hz 
> AUDIO DEVICE COUNT:  2
> SDL_VIDEO_DRIVER selected: KMSDRM
> SDL_RENDER_DRIVER selected: opengles2
> Rpi IO created
> END HostInit
> Gamepads currently attached: 1
> Controller is:  8BitDo SN30 Pro
> Will try to read rpi settings file: /home/pi/.config/Chiaki/Chiaki_rpi.conf
> Couldn't open the settings file
> Discovery Service started
> GL_VERSION  : OpenGL ES 3.1 Mesa 19.3.2
> GL_RENDERER : V3D 4.2
> Gui init done
> Gamepad added
> Controller is:  8BitDo SN30 Pro
> Discovered Host State:  ready
> Is PS5:  0
> No Hosts found in file
> Discovered Host State:  ready
> Is PS5:  0
> No Hosts found in file

Fredrum, 2023-07-30:
Hi unfortunately I don't have the bandwidth to work on this project any more.
I'd reccommend trying disconnecting one of you PS's and then restarting Chiaki and see if that helps.

On can actually have two PS's set up by using the 'remote' functionality and pretending that the secondary PS is remote even though it isn't. That lets you have two sets of setting prefs.

But of course you still need to get the basics working.