Fredrum/chiaki #38: Chiaki RPi02 Fails to Register PS4

Boozmd, 2022-09-10:
So I have an RPi 4 Model B 4GB, Running the latest version of Bullseye. I had Chiaki RPi02 running pretty smoothly on an older OS version (cant remember which) over WiFi, but decided to wipe the Pi and start with a fresh build as it was full of shit.

I have followed the steps meticulously on the new build of Chiaki RPi02 and when Registering the PS5, I get
END RegistClick
In host RegistEventCB
Registration fail

So I went through the process with Chiaki RPi01 and I am able to register the PS5, however there is sound but no video. I didn't feel like messing with it further, just wanted that conf file.

I renamed the Chiaki.conf file from Chiaki RPi01 to Chiaki_rpi.conf and ran the commands to start Chiaki RPi02, it shows the blank ? square with UNKNOWN underneath instead of NOTREG.
I click to start and get
Host Click
Status Unknown.

I'm kinda lost at this point.

GianniDPC, 2022-09-22:
Having the same issue testing on a RPi3. PS5 is not detected at all and manually setting the IP and trying to register the console results in « Registration fail ». What am I doing wrong?

Boozmd, 2022-09-22:
> Having the same issue testing on a RPi3. PS5 is not detected at all and manually setting the IP and trying to register the console results in « Registration fail ». What am I doing wrong?

Are you getting the 8-digit code to enter for registration?

GianniDPC, 2022-09-23:
> > Having the same issue testing on a RPi3. PS5 is not detected at all and manually setting the IP and trying to register the console results in « Registration fail ». What am I doing wrong?
> 
> Are you getting the 8-digit code to enter for registration?

Yes I have entered the account id and pin and after confirming these credentials I get the "Registration fail" message.

EelcoMulder, 2022-09-23:
I have the same problem, got Playstation ID and a pin but 'Registration fail'

CyberCowboy, 2022-10-20:
Same problem after new install of Bullseye and compiling RPi02.  Tried bringing my RPi01 config as well as creating a new one.

triple-it, 2022-10-25:
Hi... having the same problem. Running with Bullseye and RPI02. 
Will try a RP01 build as well...

`AccID entered:  <something>
Regist Target:  800
PIN int:  27993368
Host Addr:  10.51.1.24
END RegistClick
In  host RegistEventCB
Registration fail
`

triple-it, 2022-10-25:
RP01 registration works (but then no video) with:  ./gui/chiaki --platform eglfs --retropie

Small remark though... the host field at registration was autopopulated with an other PS5 first, instead of my discovered one, so I had to override the Host IP setting to the correct one to work.

But RP01 at least registration works... the chiaki-rpi not yet...

Fredrum, 2022-12-27:
@triple-it @CyberCowboy @EelcoMulder @GianniDPC 

Hi sorry everyone for the failing PS5 register I have not owned a PS5 so I couldn't test or fix things. Probably shouldn't have said it was working on PS5 when I didn't own one. 
Anyway I now have one in the house as I got one for my son for xmas. (yeah also somewhat for me)

I just added a fix for the PS5 register that works here for me.
So you can re-clone or do a 'git pull' and rebuild and I think it should work?

Something seems to be broken with h264 decode so
make sure you use 'h265' 
(Automatic should in theory do the same if you have an up to date RPi4b + RPiOS)

Its working for me and I get great low latency results with:

RPI4b + latest RPiOS
PS5
1080 resolution, 60 fps
h265  (or 'Automatic') codec
Ethernet to both PS5 and Pi
Wired controllers