Fredrum/chiaki #40: Send wakeup when starting stream from cli

tinus, 2023-01-26:
I have started using your rpi01 and it works very well. I like to start it from cli because my pi doesn't have a mouse attached to it. That works well too but the ps5 has to be awake for that to work. Is there any way to have chiaki send the wakeup call from the client? 

I did see a file in the source that hints at that but I couldn't figure it out.

Thank you for the great work.

Fredrum, 2023-01-27:
Hi I'll be able to try and help you more in a few days.
In the meantime,

- Did you get the cli executable in _build/cli_  ?
- 
If you run you should be able to get this help text,
https://github.com/Fredrum/chiaki/blob/250bd236dd4282aa1015de9365e63a8926e06028/cli/src/main.c#L12

tinus, 2023-01-27:
ooh, that is a separate executable. Thank you very much, it is being built now. That will help a lot :)

tinus, 2023-01-27:
It works :)
Thank you very much.

tinus, 2023-01-28:
I managed to write a udev rule that triggers when my stadia controller connects to the pi over bluetooth. That runs a script which calls chiaki-cli to wakeup the ps5, then opens the stream and then calls a last xdotool key F11 to put it in fullscreen.

I am happy :)

closed 2023-01-28 by tinus