Fredrum/chiaki #41: PS5 State Unknown

cmcgowan111, 2023-02-07:
Hi. I recently discovered this new RPI02 and managed to follow all the instructions to build this on my Pi 4b. The initial registration of my PS5 works fine and I can play for as long as I want. Once I disconnect, the state of the console changes to "Unknown" and I can't figure out how to resolve this.  The shell output after running the chiaki-rpi command shows the correct state but the GUI has the "Unknown" state.  A reboot of both the Pi and the PS5 doesn't help. Just wondering if you have seen this issue and can please help with how to resolve it. Thanks

Fredrum, 2023-02-09:
Hi I just tried running again both on desktop and on tty/cli and could not reproduce that behaviour.

What OS are you using on your RPi?

Do you use WiFi on either the PS5 or the RPi?

Do you have a folder named something like:

`/home/pi/.config/Chiaki`  

with a file named _Chiaki_rpi.conf_

cmcgowan111, 2023-02-09:
Hi,

Thanks for getting back to me.

I am using Bullseye 64bit. Both the PS5 and RPi are using an ethernet cable
and are connected to the same switch at the moment whilst I've been
testing. I had to create the /.config/Chiaki folders and the
Chiaki_rpi.conf file myself and the owner is my own user that I am running
it with.  The shell output does show that it reads the config file ok,
finds the PS5 and that it is in ready state. Once the GUI loads it had the
question mark and unknown displaying.

I have done a few clean installs and always end up with the same error
unfortunately.

Is there any way to enable some extra logging to a file maybe to help see
where it going wrong?

Many thanks for the help.

cmcgowan111, 2023-02-09:
Should have added the below in previous message. 

I know they are completely different but I have the android Chiaki and PS Play apps on my phone and my TV and these both work fine. Official remote play app works too so it just this RPi one I'm having problems with.

Fredrum, 2023-02-09:
Before we try to add some more printouts could you please paste the entirety of the shell output either here or on pastebin?
Including up to the error happens.

cmcgowan111, 2023-02-10:
Sure, once I find the SD card I have misplaced. I swapped it for another
one to install Buster and follow your RPI01 build instructions. That is
working fine and I haven't had any issues with it not detecting the PS5
state properly. The controllermap also worked great to get my Stadia
controller connected via Bluetooth working too.

Hopefully I will find the missing SD card this weekend but if not, I will
wipe the Buster one I have to install Bullseye again and get back to where
we were so we can troubleshoot further.

Thanks again for your help.

cmcgowan111, 2023-02-15:
Hi,

Sorry for the delay. Please see the shell output below.

pieraspberrypi:~$ /home/pi/chiaki-rpi/build/rpi/chiaki-rpi
Linux base Path: /home/pi/chiaki-rpi/build/rpi/chiaki-rpi
h265 Hardware Decoder Detected
SDL version linked: 2.0.20 Running in Desktop mode: 0
Disp Mode: 3840x2160 @ 60hz AUDIO DEVICE COUNT: 3
INFO: Audio device 0: bcm2835 Headphones, bcm2835 Headphones INFO: Audio
device 1: vc4-hdmi-0, MAI PCM i2s-hifi-0
INFO: Audio device 2: vc4-hdmi-1, MAI PCM i2s-hifi-0 SDL_VIDEO_DRIVER
selected: KMSDRM SDL_RENDER DRIVER selected: opengles2
INFO: Audio device 0: bcm2835 Headphones, bcm2835 Headphones
INFO: Audio device 1: uc4-hdmi-0, MAI PCM i2s-hifi-0 INFO: Audio device 2:
uc4-hdmi-1, MAI PCM i2s-hifi-0
Rpi 10 created END HostInit
Gamepads currently attached: 1 Controller is: (null)
Will try to read rpi settings file: /home/pi/.config/Chiaki/Chiaki_rpi.conf
Found settings for 1 host(s)
Discovery Service started
GL_VERSION : OpenGL ES 3.1 Mesa 20.3.5 GL_RENDERER: V3D 4.2
Gui init done Gamepad removed
Discovered Host State: standby
Is PS5: 1
Gamepad added
Controller is: PS5 Controller
Host Click
State Unknown
Host Click
State Unknown
Host Click
State Unknown
Host Click
State Unknown
Host Click
State Unknown
Host Click
State Unknown

So it looks like it finds the PS5 ok and knows it is in the "standby*
state, the GUI loads ok but it shows like in the screenshot attached.

Fredrum, 2023-02-15:
Hi thanks for that!
I should have some time during the weekend to take a look.
Can't promise I'll figure it out though.

Fredrum, 2023-02-19:
Hmm your printout is missing a line after `Is PS5:  1` where it should say that it has correctly identified your particular PS5 from whats stored in the config file. You should get a line saying 'Discovery ID match":
```
Discovered Host State:  standby
Is PS5:  1
Discovery ID match
```

You said earlier that you had to create the 'Chiaki' config folder and config file yourself which makes it seem something could have gone wrong there.

This is how my config file looks:
(I have changed a few letters/numbers here and there for security)
```
version: 1

regist_hosts:
  - host : 1
    isPS5 : 1
    nick : PS5-512
    id : 70862A1FC2BA
    remote_ip : 0.0.0.0
    rp_key : ynFLFv05xkNchgskLFlCLA==
    regist : 324ea7ee
    session : 
      decoder : automatic
      codec : h265
      resolution : 1080
      fps : 60
      audio_device : Built-in Audio Digital Stereo (HDMI)
```

Does your file have all the same data?

cmcgowan111, 2023-02-20:
Hi. This is what my config file looks like. I also changed some of the
character. I assume the "ID" should be the MAC address of the PS5 as that
is what it is.

***@***.***:~ $ cat /home/pi/.config/Chiaki/Chiaki_rpi.conf

version: 1

regist_hosts:
  - host : 1
    isPS5 : 1
    nick : PS5-312
    id : 00F431234255
    remote_ip : 0.0.0.0
    rp_key : JMK9POKjtElGgDNfIVfibQ==
    regist : 52f2b53a
    session :
      decoder : automatic
      codec : automatic
      resolution : 1080
      fps : 60
      audio_device : vc4-hdmi-0, MAI PCM i2s-hifi-0
***@***.***:~ $

When I created the Chiaki_rpi.config file I left it blank and made sure the
pi user was the owner. The rest was added once I started the app and
registered the console the first time.

Fredrum, 2023-03-03:
I just noticed something...could you tell me what's the output when you do,

`uname -a`

To get your kernel version?

cmcgowan111, 2023-03-04:
***@***.***:~ $ uname -a
Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023
aarch64 GNU/Linux

Fredrum, 2023-03-05:
Thank you for that.
I upgraded my Pi4 to the new kernel 6.1 and now I get some problem that Chiaki won't discover my PS5 it its in standby mode. Chiaki now reports 'unknown' until I have manually turned on my PS5.

But unfortunately that's not the same as your issue then. 
Shame as I need to be able to replicate your problem to be able to fix it.

Hm ok. Sorry I don't have any more leads on this atm.

cmcgowan111, 2023-03-05:
No worries. Thanks very much for helping so far.

I'll do some testing to see if I can get it to work like yours does now. I
can live with having to use my phone app to switch it on before launching
chiaki on the Pi4.