Fredrum/chiaki #43: Chiaki crashes on pi4 when ps5 dashboard is shown

pepe6414, 2023-02-21:
Hi!
Chiaki version rpi02 is crashing after the ps dashboard is displayed on the screen. Looking at the console, I noticed this error:
AVERROR(EAGAIN): [hevc @ 0xf925a0] Hwaccel V4L2 HEVC stateless V4; devices: /dev/media0,/dev/video19; buffers: src DMABuf, dst DMABuf; swfmt=rpi4_8

Googling the error it seems to be related to a hevc codec failing to map the frames. Any idea on how to fix it?

I followed the instructions to modify the /boot/config.txt and tried different alternatives/configurations but at the end of the day it fails.

Fredrum, 2023-02-21:
Is this consistently happening for you?
Every time you start?

I currently don't do any edits in /boot/config.txt or change any memory settings and it works pretty well here for me.

Are you using Ethernet on both sides or are you using wifi for PS5 or RPi?

pepe6414, 2023-02-21:
It is is happening since the first I register the console, I am just able to see my ps5 dashboard for a couple of seconds and chiaki chrashes. I can repeat the issue everytime I try to connect. 
Regarding the /boot/config.txt I followed the instructions described on forks rpi01 and rpi02 but still the same issue.
Today I tried modifying the resolution and frames from chiaki UI to the lowest, from the console I was able to ready failed to create the frame POC

Fredrum, 2023-02-21:
Are you using Ethernet on both sides or are you using wifi for PS5 or RPi?

What version of Raspberry Pi do you have?

What operating system are you using on the RPi?

pepe6414, 2023-02-21:
I have Raspberry pi Os legacy 32 bits over a pi4b
Lastest firmware update launched on September 2022

Fredrum, 2023-02-22:
Could you please link to the image you used "Raspberry pi Os legacy 32 bits".
If I cannot reproduce I don't stand much chance of fixing anything as I don't have the time to just randomly test different things.

pepe6414, 2023-02-22:
Hello!
Sure. Please see below:

Raspberry OS version
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye

Firmware version:
$ sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 2023 05:40:52 PM UTC (1673458852)
    LATEST: Wed 11 Jan 2023 05:40:52 PM UTC (1673458852)

Fredrum, 2023-02-22:
Thanks for that.
I was actually thinking do you have the download link for the image?

pepe6414, 2023-02-22:
I installed it directly from bootloader but, looking at Raspberry Os page this is the one I got: https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2022-09-26/2022-09-22-raspios-bullseye-armhf-full.img.xz

Fredrum, 2023-02-26:
Ooook I downloaded and flashed that image and got the same crashes as you did.
BUT I then also ran the latest OS updates (as prompted but the icon in the top menu bar) on the card that WAS working for me.
After that update it started crashing like you report. (and another new user issue)

So it seems like something in the new RPiOS update makes Chiaki crash.

I can run in 720p though without crashing.

Not sure if or when I'll be able to fix this. This software is not a life priority for me just something I do now and then for fun.
And chasing linux updates that break stuff is not fun. Just mentioning so you understand this might take a while to get fixed.

pepe6414, 2023-02-26:
Thanks for the update. I am happy that you were able to reproduce this issue. 
Reinstalling the Os image is not an alternative since I will loose some data I cannot transfer to another Os image. 
Chiaki is the only alternative I have found so far to play, if you know about any other alternative, I really appreciate your feedback.

Fredrum, 2023-03-14:
It seems to me that this happens on the 32bit Raspberry OS version but not on the 64bit one.

pepe6414, 2023-03-14:
Hi Fredrum!
Thank you so much for the updates, I really appreciate it you did not dropped/put at a side this issue.
After your update I did a backup of my old 32bit Rasp OS and installed a 64bit Rasp OS, after installing all the updates/upgrades I managed to install chiaki on my pi4 and it is now working
It is throwing some errors please see the screeshots but, goog thing is it is not crashing. I am able to play a game and enter to the different options:
[image]
[image]

One thing I noticed at the begining is that the dualsense was not working properly and the scree got frozen couple of times, so what I did, is to install the dualse sense driver for linux and not I do no see those errors

Fredrum, 2023-03-15:
Thanks for the report! 
I just tried running for a while on my new flashed 64bit card both on desktop and on tty and am not getting sny of those error messages. But I feel I recognize them from the problems on 32bit so maybe they are part of the same 'weakness'. 

For the DualSense, the controller input is done through SDL and the version I pull in my install script (2.0.20) is not that new so maybe that PS5 controller didn't have good support a that time. If you do it again you could try installing a newer SDL version and see if it helps. I'll try at some point too as it would be good to know if I should upgrade that script.

Fredrum, 2023-03-15:
I found a thing that was causing the crashes on 32bit and made a 'fix'.
Fix in quotation marks because I don't understand what's changed and why that thing now cause crashes.
It was in the audio code and not in the video part.
Not sure if its going to get rid of your new error messages though?

pepe6414, 2023-03-15:
Are you pushing the update/fix to the 64bits?
I can help testing it out if you do

Fredrum, 2023-03-15:
No this is just a fix to my Chiaki code which runs the same on both 32 and 64 bit.
So yes please if you can re git clone or refresh and see if those error messages disappear that would be very interesting!

I got no reply on the Raspi forum so not sure what actual change caused this, or why this error happens at all.
I would have to spend mor etime trying to understand whats really going on but I won't do that. I'm just content that it seems to run on 32 bit too now.

pepe6414, 2023-03-15:
Hi Fredrum!
Just removed the previous chiaki version and git clone the new code. Now I have a black screen and not able to see any image, but I cannot see error on the console:
[image]
[image]

Fredrum, 2023-03-15:
oh god. :P 
I see you have no Audio device selected in the gui!?
Can you please try and get something selected there and run again?

pepe6414, 2023-03-15:
Blackmagic haha I picked up my HDMI port as audio and I can see an image now.
Unfortunately as expected, the errors are still there:

[image]

Good thing is it is playable and it is not crashing as before

Fredrum, 2023-03-15:
Ok cool thanks for testing. Obviously its not good that it crashes without the audio device it should still just go ahead and do the video nonetheless so I'll have to look at that.

Shame that there's still error messages on the video decode. I wonder why I don't get them?
What's your networking situation?

pepe6414, 2023-03-15:
I am connected to the same network with a Ethernet cable in both. My download/upload speed is +-80 MB

How Can I run the tty/cli screen?

Fredrum, 2023-03-15:
Ok your network setup seems as solid as mine so don't know then why its different.

To get to tty from the desktop you do,
Ctrl+Alt+F1       (for tty1 you can also use F2, F3)

To get back to Desktop you can do,
Ctrl+Alt+F7      (Maybe sometimes F8?)

Note that what tty's are used is slightly different on differnt OSs like Ubuntu would be a little differnt but generally similar.

Fredrum, 2023-03-15:
Btw do you get those error mesages on each frame?
Like is there a constant fast stream of them?
Or do you get new ones just on occational frames?

pepe6414, 2023-03-15:
I tried the tty mode. The image quality improved a bit but the latency/lag was the same per my feeling.
Yes. The errors seems to be each frame poc3, poc4
I start getting the error as soon as I start moving the cursor, it occasionally happens without interacting with the options/screen

Fredrum, 2023-03-15:
hmok so no errors before you move the mouse?
One problem with streams of errors is that IO slows the process down and if there's enough printouts it will slow down the program. You'll get worse latency.
But again, since I can't reproduce, I don't get those errors, I'm not able to try and fix.
But at least it sounds like its useable for you.

pepe6414, 2023-03-15:
No errors before moving the mouse/cursor 90% of times. It could be the buffer cannot handle/map the printouts or the pointer cannot find a free space to map it, but it is playable. The sensation/feeling is like when the have a bad network connection/lag.
I am going to test it for a couple of days and share my comments

Fredrum, 2023-03-18:
Ive made a few fixed with regards to sound device selection and audio crashes over the last couple of days so you might want to re-install when you get back to this.

pepe6414, 2023-03-20:
Hello!
Just re-installed the version, the first change I noticed is by default it selects an audio output so you do not get the black screen because you run it without picking one, second I noticed now on the console I can see a warning "Audio Buffer Large Warning".
I did not see the previous errors at the first sight but, unfortunately they appeared one more time :( but you can play, the app do not crashes so I am pretty happy. Thank you so much one more time:

[image]