Fredrum/chiaki #44: Crashes as soon as I connect to PS5

RobertAdams313, 2023-02-25:
I'm not sure if it's because i'm using a Raspberry Pi 3 and not the 4 but I followed the instructions but even if I switch codecs it just crashes as soon as it connects. I can see the stream for a split second, the PS5 will say remote play started and then will say it disconnected. 

I'm using the latest OS from the imager tool here. https://www.raspberrypi.com/software/

Is the 3 just not supported?

Fredrum, 2023-02-26:
Hi Probably the same as the other issue,

https://github.com/Fredrum/chiaki/issues/43#issuecomment-1445253593

Fredrum, 2023-03-15:
I made a fix yeasterday that seems to make it work again on 32 bit RPiOS.
For me 64bit was working and not crashing.

You should be able to re-install a fresh copy and see if it fixes this for you too.

Fredrum, 2023-03-16:
Actually could you please double check that your audio device selector is not in blank/empty mode?

RobertAdams313, 2023-03-16:
> Actually could you please double check that your audio device selector is not in blank/empty mode?

I'll check this evening to see but this is a possibility because my monitor doesn't have speakers and I didn't connect headphones or anything to it at that time.

Fredrum, 2023-03-16:
Obvs my code is bad if that's what happens but you could try for now as It'll be a bunch more days before I have any time to get back to this.

Fredrum, 2023-03-18:
Iv'e made a few small fixes over the last two days with regards to sound crashes so you might want to re git clone and build the latest when you get back to this.

RobertAdams313, 2023-03-20:
Sorry ADHD has been kicking my ass when it comes to hooking it back up to test it out but I finally did it. Unfortunately it's still crashing for me. Not sure if maybe I did something wrong. 

This is what it says in the terminal. 

`Linux base Path:  /home/pi/chiaki-rpi/build/rpi/chiaki-rpi��~��u��0
WARNING:  h265 Decoder Not Detected
SDL version linked: 2.0.20
Running in Desktop mode:  1
Disp Mode: 1920x1080 @ 60hz 
AUDIO DEVICE COUNT:  3
INFO: Audio device 0: Built-in Audio Analog Stereo
INFO: Audio device 1: Built-in Audio Digital Stereo (HDMI)
INFO: Audio device 2: DualSense Wireless Controller Analog Surround 4.0
SDL_VIDEO_DRIVER selected: x11
SDL_RENDER_DRIVER selected: opengles2
INFO: Audio device 0: Built-in Audio Analog Stereo
INFO: Audio device 1: Built-in Audio Digital Stereo (HDMI)
INFO: Audio device 2: DualSense Wireless Controller Analog Surround 4.0
Rpi IO created
END HostInit
Gamepads currently attached: 1
Controller is:  PS5 Controller
Will try to read rpi settings file: /home/pi/.config/Chiaki/Chiaki_rpi.conf
Found settings for 1 host(s)
Discovery Service started
GL_VERSION  : OpenGL ES 2.0 Mesa 20.3.5
GL_RENDERER : VC4 V3D 2.1
Discovered Host State:  ready
Is PS5:  1
Discovery ID match
Gui init done
Gamepad added
Controller is:  PS5 Controller
Host Click
Starting Play session!
SDL drm_fd:  35651612
[h264_v4l2m2m @ 0x1d51290] level=-99
[h264_v4l2m2m @ 0x1d51290] Using device /dev/video10
[h264_v4l2m2m @ 0x1d51290] driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[h264_v4l2m2m @ 0x1d51290] requesting formats: output=H264 capture=YU12
Actual Fmt: 181
Finish InitFFmpeg
Requesting codec: h264
END chiaki session start
Init Audio CB
ConnectionEventCB() CHIAKI_EVENT_CONNECTED
Starting Play Thread
Connected Setsu Touchpad Device /dev/input/event13
Connected Setsu Motion Device /dev/input/event12
AVERROR(EAGAIN)
AVERROR(EAGAIN)
AVERROR(EAGAIN)
AVERROR(EAGAIN)
AVERROR(EAGAIN)
AVERROR(EAGAIN)
AVERROR(EAGAIN)
Error return in send_packet
Segmentation fault
`

RobertAdams313, 2023-03-20:
Ok I'm an idiot. I forgot to connect the Pi to my WiFi. I connected to my WiFi and it doesn't crash anymore but the quality of the stream artifacts like crazy and overall lags but I'm assuming that's a network/weak pi thing rather than the program.

RobertAdams313, 2023-03-20:
Well I spoke too soon. I tried opening it again and it just keeps crashing out when I connect. I've tried with AirPods connected just in case it was an audio thing. I can hear it connect to the PS but then it just closes.

It's throwing that AVERROR (EAGAIN) 
and then [E] FEC failed
Error return in send_packet
Segmentation fault

Fredrum, 2023-03-20:
Hi no worries baout taking time to re-try. I'm in no hurry! :)
So you have an Rpi3 right? I don't have one so can't try to reproduce but it did work in the past when I tried.
Its not very good in Desktop mode as I remember though.

Did you try in tty/cli?

You can get there from desktop by doing,
Ctrl+Alt+F1,  then back to Desktop with Ctrl+Alt+F7.

But it might not fix the problems possibly.
I can't remember if I restrict the resolutoin but I think Rpi3 might only support 720p is that what youre testing with?

Fredrum, 2023-03-21:
I just tried again with my Pi Zero 2 W, over wifi, and it works on that. (but pretty sluggish over the wifi)
That's as close as I can get to a Pi3. Not sure what to suggest.