Fredrum/chiaki #45: Start chiaki with a service

AndSchroeder, 2023-03-16:
Can you add an instruction on how to create a service or a way to start chiaki after startup of the pi?

Fredrum, 2023-03-17:
Actually its probably be if you just google it and try. I'm not going to have much time to do this and its sort of a separate thing.
But please if you get a good solution post it here for others maybe?

brantai, 2023-03-17:
I use the following .service file in my user-level systemd directory to start it:

```
[Unit]
Description=Chiaki
After=network.target

[Service]
ExecStart=/home/brantai/chiaki-rpi/build-release/rpi/chiaki-rpi 2>&1
Restart=always

[Install]
WantedBy=default.target
```

Replace the ExecStart with wherever you've built your chiaki-rpi binary.

Fredrum, 2023-03-17:
I remember in my version I made a bash script instead of calling the executable straight from the service.
I then added something like a 'sleep 3' in which allowed me to cancel out inbetween restarts if I needed to do some fixes.
Otherwise I got stuck in a restart loop which was a bit tricky to break out of.

AndSchroeder, 2023-03-19:
Hey I was not able to start it with a service, but I found another approach.
I created an autostart file to start a script in the gui.

first of all I created the bash script for starting chiaki directly in my home dir (start-chiaki.sh):

```
#! /bin/bash
sleep 5
chiaki-rpi/build/rpi/chiaki-rpi 2>&1
```

Additionally I added a gui autostart file

`sudo nano /etc/xdg/autostart/psremote.desktop`

After that I tun the bash script to start chiaki

```
[Desktop Entry]
Exec=/home/username/start-chiaki.sh
```
Than i restarted with `sudo reboot` and chiaki started as well.

Now I only have to find a way to start it in desktop mode. Maybe another section in the wiki about autostart would be helpful.

closed 2023-03-19 by AndSchroeder