Fredrum/chiaki #49: make: *** [Makefile:156: all] Error 2

traelor55, 2023-12-05:
Ive followed the guide step by step but I can't seem to get this to install correctly. Everything goes smoothly up until the end

When running the "make -j3" command I get the following error

```
make[2]: *** [rpi/CMakeFiles/chiaki-rpi.dir/build.make:90: rpi/CMakeFiles/chiaki-rpi.dir/src/io.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:378: rpi/CMakeFiles/chiaki-rpi.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

Has anyone else experienced this error and found a work around? Im on a Raspberry Pi B4?
Is this possibly a permissions issue?

traelor55, 2023-12-05:
So for anyone who doesn't read the fine print like me. You need to make sure you are on the correct version of Pi OS. I was running Bookworm but switching to Bullseye fixed the issue

closed 2023-12-05 by traelor55

mroshaw, 2024-12-27:
For info, I've got Chiaki working on a Pi 4B with a fresh 64-bit Bookworm OS using changes that death-droid shared in a pull request. In the file `rpi\src\io.cpp` Replace line 445 (`AVCodec *av_codec;`) with:
```cpp
	#if LIBAVCODEC_VERSION_MAJOR < 59
		AVCodec *av_codec;
	else
		const AVCodec *av_codec;
	#endif```
Build again with `make -j3`. Worked for me with Wayland and X-11.